oscam-emu/oscam-emu #103: Hunk #4 FAILED at 333

JanGruuthuse, 2015-01-12:
with changeset [10170] and simplebuild

apply patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
Hunk #1 succeeded at 1 with fuzz 2.
patching file config.sh
Hunk #4 FAILED at 333.
1 out of 5 hunks FAILED -- saving rejects to file config.sh.rej
patching file cscrypt/des.c

JanGruuthuse, 2015-01-12:
config.sh.rej
=========

--- config.sh	(working copy)
+++ config.sh	(revision 10001)
@@ -331,9 +329,9 @@
 	not_have_flag USE_LIBCRYPTO && echo "CONFIG_LIB_AES=y" || echo "# CONFIG_LIB_AES=n"
 	enabled MODULE_CCCAM && echo "CONFIG_LIB_RC6=y" || echo "# CONFIG_LIB_RC6=n"
 	not_have_flag USE_LIBCRYPTO && enabled MODULE_CCCAM && echo "CONFIG_LIB_SHA1=y" || echo "# CONFIG_LIB_SHA1=n"
-	enabled_any MODULE_NEWCAMD READER_DRE MODULE_SCAM WITH_EMU && echo "CONFIG_LIB_DES=y" || echo "# CONFIG_LIB_DES=n"
-	enabled_any MODULE_CCCAM READER_NAGRA WITH_EMU && echo "CONFIG_LIB_IDEA=y" || echo "# CONFIG_LIB_IDEA=n"
-	not_have_flag USE_LIBCRYPTO && enabled_any READER_CONAX READER_CRYPTOWORKS READER_NAGRA WITH_EMU && echo "CONFIG_LIB_BIGNUM=y" || echo "# CONFIG_LIB_BIGNUM=n"
+	enabled_any MODULE_NEWCAMD READER_DRE MODULE_SCAM && echo "CONFIG_LIB_DES=y" || echo "# CONFIG_LIB_DES=n"
+	enabled_any MODULE_CCCAM READER_NAGRA && echo "CONFIG_LIB_IDEA=y" || echo "# CONFIG_LIB_IDEA=n"
+	not_have_flag USE_LIBCRYPTO && enabled_any READER_CONAX READER_CRYPTOWORKS READER_NAGRA && echo "CONFIG_LIB_BIGNUM=y" || echo "# CONFIG_LIB_BIGNUM=n"
 }
 
 make_config_c() {

allemallachen, 2015-01-13:
Look at streamboard ... there is a Patch for r10184 ...

jahntheman, 2015-01-21:
Fixed

closed 2015-01-21 by jahntheman