oscam-emu/oscam-emu #104: Can't build actual SVN with oscam-emu Patch

al-x83, 2015-01-18:
Here's the build-log:

+-------------------------------------------------------------------------------
| OSCam ver: 1.20-unstable_svn-emu- rev: 10225 target: x86_64-linux-gnu
| Tools:
|  CROSS    = 
|  CC       = gcc
| Settings:
|  CONF_DIR = /usr/local/etc
|  CC_OPTS  = -O2 -ggdb -pipe -ffunction-sections -fdata-sections
|  CC_WARN  = -W -Wall -Wshadow -Wredundant-decls -Wstrict-prototypes -Wold-style-definition
|  CFLAGS   = 
|  LDFLAGS  = -Wl,--gc-sections
|  LIBS     = -lpthread -ldl
|  UseFlags = 
| Config:
|  Addons   : WEBIF WEBIF_LIVELOG WEBIF_JQUERY TOUCH HAVE_DVBAPI IRDETO_GUESSING CS_ANTICASC WITH_DEBUG MODULE_MONITOR WITH_LB CS_CACHEEX CW_CYCLE_CHECK LCDSUPPORT
|  Protocols: CAMD33 CAMD35 CAMD35_TCP NEWCAMD CCCAM CCCSHARE GBOX RADEGAST SERIAL CONSTCW PANDORA GHTTP
|  Readers  : NAGRA IRDETO CONAX CRYPTOWORKS SECA VIACCESS VIDEOGUARD DRE TONGFANG BULCRYPT GRIFFIN DGCRYPT
|  CardRdrs : PHOENIX INTERNAL SC8IN1 MP35 SMARGO DB2COM STAPI STINGER  
|  Compiler : gcc (Debian 4.4.5-8) 4.4.5
|  Binary   : Distribution/oscam-1.20-unstable_svn-emu-10225-x86_64-linux-gnu
+-------------------------------------------------------------------------------
CC	cscrypt/des.c
CC	cscrypt/md5.c
CC	csctapi/atr.c
CC	csctapi/icc_async.c
CC	csctapi/io_serial.c
CC	csctapi/protocol_t0.c
CC	csctapi/protocol_t1.c
CC	csctapi/ifd_db2com.c
CC	csctapi/ifd_mp35.c
CC	csctapi/ifd_phoenix.c
CC	csctapi/ifd_sc8in1.c
CC	csctapi/ifd_sci.c
CC	csctapi/ifd_smargo.c
CC	csctapi/ifd_stinger.c
CC	module-anticasc.c
CC	module-cacheex.c
CC	module-camd33.c
CC	module-camd35.c
CC	module-cccam.c
CC	module-cccshare.c
CC	module-constcw.c
CC	module-csp.c
CC	module-cw-cycle-check.c
CC	module-dvbapi-chancache.c
CC	module-dvbapi.c
CC	module-gbox-helper.c
CC	module-gbox-sms.c
CC	module-gbox.c
CC	module-ird-guess.c
CC	module-lcd.c
CC	module-monitor.c
CC	module-newcamd.c
CC	module-pandora.c
CC	module-ghttp.c
CC	module-radegast.c
CC	module-serial.c
CC	module-stat.c
CC	module-webif-lib.c
CC	module-webif-tpl.c
CC	module-webif.c
CC	webif/pages.c
CC	reader-common.c
CC	reader-bulcrypt.c
CC	reader-conax.c
CC	reader-cryptoworks.c
CC	reader-dgcrypt.c
CC	reader-dre.c
CC	reader-griffin.c
CC	reader-irdeto.c
CC	reader-nagra.c
CC	reader-seca.c
CC	reader-tongfang.c
CC	reader-viaccess.c
CC	reader-videoguard-common.c
CC	reader-videoguard1.c
CC	reader-videoguard12.c
CC	reader-videoguard2.c
CC	oscam-aes.c
CC	oscam-cache.c
CC	oscam-chk.c
CC	oscam-client.c
CC	oscam-conf.c
CC	oscam-conf-chk.c
CC	oscam-conf-mk.c
CC	oscam-config-account.c
CC	oscam-config-global.c
CC	oscam-config-reader.c
CC	oscam-config.c
CC	oscam-ecm.c
CC	oscam-emm.c
CC	oscam-failban.c
CC	oscam-files.c
CC	oscam-garbage.c
CC	oscam-lock.c
CC	oscam-log.c
CC	oscam-log-reader.c
CC	oscam-net.c
CC	oscam-llist.c
CC	oscam-reader.c
CC	oscam-simples.c
CC	oscam-string.c
CC	oscam-time.c
CC	oscam-work.c
CC	oscam.c
LINK	Distribution/oscam-1.20-unstable_svn-emu-10225-x86_64-linux-gnu.debug
build/x86_64-linux-gnu/reader-common.o: In function `reader_get_cardsystem':
/home/administrator/OScam-compile/oscam-emu/reader-common.c:136: undefined reference to `reader_emu'
build/x86_64-linux-gnu/oscam.o: In function `main':
/home/administrator/OScam-compile/oscam-emu/oscam.c:1396: undefined reference to `cardreader_emu'
/home/administrator/OScam-compile/oscam-emu/oscam.c:1467: undefined reference to `add_emu_reader'
collect2: ld returned 1 exit status
make[1]: *** [Distribution/oscam-1.20-unstable_svn-emu-10225-x86_64-linux-gnu.debug] Fehler 1
make: *** [all] Fehler 2

valberbox, 2015-01-19:
In my system issue compares starting from svn 10214

jackuzzy, 2015-01-20:
Same here using latest nightly revision.

rodolphedj, 2015-01-21:
Hi,

pending update,
but you can open config.sh.rej and apply manually the delta (WITH EMU).
no issue with oscam 10245 and manual correction.

jahntheman, 2015-01-21:
Fixed

closed 2015-01-21 by jahntheman