oscam-emu/oscam-emu #107: failed to build oscam-svn10255-openpli40-webif-libcrypto-oscam-emu-patched

JanGruuthuse, 2015-01-21:
CC	reader-videoguard-common.c
reader-viaccess.c: In function 'viaccess_get_emm_filter':
reader-viaccess.c:1341:34: error: 'struct s_reader' has no member named 'via_emm_global'
   if((rdr->typ == R_CAMD35 && rdr->via_emm_global == 1) || rdr->typ == R_EMU)
                                  ^
reader-viaccess.c:1352:34: error: 'struct s_reader' has no member named 'via_emm_global'
   if((rdr->typ == R_CAMD35 && rdr->via_emm_global == 1) || rdr->typ == R_EMU)
                                  ^
make[1]: *** [build/mipsel-openpli40-linux-gnu/reader-viaccess.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

jahntheman, 2015-01-21:
happend when compiling without camd35. is fixed now.
thanks

closed 2015-01-21 by jahntheman

JanGruuthuse, 2015-01-21:
your welcome: build is ok now: up and running. thanks