oscam-emu/oscam-emu #111: biss

morische, 2015-01-21:
Trying to use oscam-emu for biss

It seems it doesn´t try to clear these channels

2015/01/22 00:04:05 598078 9F 80 32 28 03 44 84 09 00 13 01 81 08 0C FD AC
2015/01/22 00:04:05 598078 E7 10 00 00 01 82 02 01 00 84 02 19 C8 02 19 C9
2015/01/22 00:04:05 598078 00 00 06 19 CD 00 00 03 19 CA 00 00
2015/01/22 00:04:05 598078 c (dvbapi) capmt:
2015/01/22 00:04:05 598078 03 44 84 09 00 13 01 81 08 0C FD AC E7 10 00 00
2015/01/22 00:04:05 598078 01 82 02 01 00 84 02 19 C8 02 19 C9 00 00 06 19
2015/01/22 00:04:05 598078 CD 00 00 03 19 CA 00 00
2015/01/22 00:04:05 598078 c (dvbapi) Receiver sends PMT command 3 for channel 4484
2015/01/22 00:04:05 598078 c (dvbapi) Receiver wants to demux srvid 4484 on adapter 0000 camask 0001 index 0000 pmtpid 19C8
2015/01/22 00:04:05 598078 c (dvbapi) Found pmt type: 81 length: 8 (assuming enigma private descriptor: namespace cfdace7 tsid 1000 onid 01)
2015/01/22 00:04:05 598078 c (dvbapi) Found pmt type: 82 length: 2
2015/01/22 00:04:05 598078 c (dvbapi) Found pmt type: 84 length: 2
2015/01/22 00:04:05 598078 c (dvbapi) Found stream type: 02 pid: 19c9 length: 0
2015/01/22 00:04:05 598078 c (dvbapi) Found stream type: 06 pid: 19cd length: 0
2015/01/22 00:04:05 598078 c (dvbapi) Found stream type: 03 pid: 19ca length: 0
2015/01/22 00:04:05 598078 c (dvbapi) Found 0 ECMpids and 2 STREAMpids in PMT
2015/01/22 00:04:05 598078 c (dvbapi) Receiver wants to demux srvid 4484 on adapter 0000 camask 0001 index 0000 pmtpid 19C8
2015/01/22 00:04:05 598078 c (dvbapi) New program number: 4484 (FFFE:4484 unknown) [pmt_list_management 3]
2015/01/22 00:04:05 598078 c (dvbapi) Demuxer #0 demux_index: 0 ca_mask: 01 program_info_length: 19 ca_pmt_list_management 03

jahntheman, 2015-01-22:
Please open a ticket here http://www.streamboard.tv/oscam/timeline
This should be fixed in oscam.

morische, 2015-01-22:
Hello,

Thanks for your reply.

What is problem in oscam? I mean.. I don´t know what´s the bug.

The reason is that oscam is detecting this channel as FFFE (FTA) not an encrypted one?

Sorry for my ignorance.

LePouSAT, 2015-01-22:
FTA = Free to Air = Non Crypted Channel ...

morische, 2015-01-22:
Hello,

I know FTA is not crypted.

I´m wondering if this is the reason of the problem (oscam is detecting a crypted channel as a FTA).

Regards

jahntheman, 2015-01-31:
fixed in oscam

closed 2015-01-31 by jahntheman

alfonso990, 2015-04-02:
any solution fro 0004 sly it card?