oscam-emu/oscam-emu #131: failed against svn10577

JanGruuthuse, 2015-02-22:
CC	oscam.c
CONF	build/mipsel-openpli40-linux-gnu/config.c
oscam.c: In function 'main':
oscam.c:1539:3: error: 'cardreader_emu' undeclared (first use in this function)
   cardreader_emu,
   ^
oscam.c:1539:3: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/mipsel-openpli40-linux-gnu/oscam.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

------
/opt/simplebuild/oscam-svn/csctapi/**cardreaders.h.rej**

--- csctapi/cardreaders.h	(working copy)
+++ csctapi/cardreaders.h	(revision 10569)
@@ -13,6 +13,5 @@
 void cardreader_smartreader(struct s_cardreader *crdr);
 void cardreader_stapi(struct s_cardreader *crdr);
 void cardreader_stinger(struct s_cardreader *crdr);
-void cardreader_emu(struct s_cardreader *crdr);
 
 #endif

Bigsailor69, 2015-02-22:
This patch (full version of oscam-emu.patch)should fix that error:
http://www.streamboard.tv/wbb2/attachment.php?attachmentid=58139

p1ngb4ck, 2015-02-22:
the mentioned patch makes my x86_64 version crash instantly on startup ...

jahntheman, 2015-02-22:
fixed

closed 2015-02-22 by jahntheman

JanGruuthuse, 2015-02-22:
thanks: up and running

p1ngb4ck, 2015-02-22:
<3. just confirming that it compiles without warnings again and runs smoothly =) Thanks !