oscam-emu/oscam-emu #140: Pvu

nicovil, 2015-04-18:
I checked out this code and noted it has the powervu algorithm implemented on it.

Is there some way to configure pvu keys on softcam.keys?

cccamcccc, 2015-04-19:
I need to have NDS1 INS54 support

nicovil, 2015-04-20:
Nds1 ins54 is needed for powervu? Isn't it for videoguard?

cccamcccc, 2015-04-20:
Yes, he is VideoGuard.
Reader initialization failed
CW encryption.

nicovil, 2015-04-20:
Buy PowerVU is another encryption different from Videoguard.
Recently colibrí has released a brute force hack paper for PowerVu and it would be nice if oscam-emu supports PowerVU since it is now possible to get the manager keys and then retrieve the ECM keys in real time from processing the EMM, and then decrypt all ECM with the algorithm you implemented on oscam-emu (at least I think this).

cccamcccc, 2015-04-20:
Hello, my friend, thank you for your words of reply,
PowerVU I know some, but I heard only the C-band can be cracked,
I have here a KU-band, if you can tell me to get back to a receiver test available to you

nicovil, 2015-04-20:
Hi!

As you said, powerVU is used for C-Band, but here in south america (and seems like north america too) there are some providers which can be retrieved with a 1.2mts dish and a Cband lnb. I have an enigma 2 receiver here and I'm able to retrieve a lot of channels without problem... now I'm looking for some software to brute force the keys (here are a lot of info, but any software was released yet: http://www.satellitetesters.com).

cccamcccc, 2015-04-20:
hi friends,
I also just learned about this OSCAM EMU,
I think you're great.
I think if you write some simple code I need should be no problem,
I'm looking for people to help me write the code, you have any good suggestions?
Or a good friend? Here it can be attached to email or skype?
Because I came, not just paste them. Ask for your help.
I am always concerned about the KU band PowerVu crack.

iptvfta, 2015-06-07:
powervu patch can do same like biss to dvb box?  now only openbox team and strong team have added this patch to their box

cccamcccc, 2015-06-07:
emuOSCAM supports KU band break it?
How I need to do?
Who can teach me.

iptvfta, 2015-06-08:
i also searching who cna help me add powervu patch to my dvb box like biss softcam function

jahntheman, 2015-07-08:
pvu has not finished yet.  it is working now,

closed 2015-07-08 by jahntheman

cccamcccc, 2015-07-09:
I want to know now was emuOSCAM can be used in X86 server? Share to my partner?

iptvfta, 2015-07-30:
powervu can working by normal dvb box now?   i still only got info by google , told only startsat, strong and openbox decoder have powervu patch function

iptvfta, 2015-10-14:
who have powervu source code of NVK78326SE/? i can payment any time. cherrs