oscam-emu/oscam-emu #154: Does not work oscam-emu patch of v10919

commando2015, 2015-07-31:
does not work oscam-emu patch of v10919
http://i.imgur.com/Pqu0wS3.jpg

oscam-emu not supported here

that is supported by 10918 and now you do not
but you made a mistake and correct make

10918 OK support oscam-emu (oscam-svn = oscam-emu.patch)
10919 NO support oscam-emu

http://i.imgur.com/MONCTNc.jpg

10900-10918 OK
http://i.imgur.com/3n0WOQU.jpg

Changed 105 minutes ago by Jan Gruuthuse 
scam-emu patch is external work ... ask for support in forum ... not ticket system.
oscam-emu patch: ​https://github.com/oscam-emu/oscam-emu/commits/master

commando2015, 2015-08-02:
918 OK support oscam-emu (oscam-svn = oscam-emu.patch)

OR

ERROR LOG
failed to build oscam-svn 10936-mipsoe20-webif-oscam-emu-patched

apply patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
Hunk #4 FAILED at 344.
1 out of 5 hunks FAILED -- saving rejects to file config.sh.rej
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
Hunk #1 succeeded at 421 (offset 17 lines).
Hunk #2 succeeded at 835 (offset 17 lines).
Hunk #3 succeeded at 965 (offset 17 lines).
Hunk #4 succeeded at 1627 (offset 27 lines).
Hunk #5 succeeded at 2085 (offset 39 lines).
Hunk #6 succeeded at 2258 (offset 42 lines).
patching file Makefile
patching file module-dvbapi.c
Hunk #1 succeeded at 1209 (offset 56 lines).
Hunk #2 succeeded at 1236 (offset 56 lines).
Hunk #3 succeeded at 3825 (offset 58 lines).
Hunk #4 succeeded at 3851 (offset 58 lines).
Hunk #5 succeeded at 5191 (offset 104 lines).
patching file module-dvbapi.h
Hunk #1 succeeded at 98 (offset 1 line).
patching file module-emulator-dre2overcrypt.c
patching file module-emulator-dre2overcrypt.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-st20.c
patching file module-emulator-st20.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
Hunk #2 FAILED at 120.
Hunk #3 succeeded at 939 (offset 2 lines).
Hunk #4 succeeded at 1365 (offset 2 lines).
Hunk #5 succeeded at 1806 (offset 2 lines).
Hunk #6 succeeded at 2363 (offset 6 lines).
Hunk #7 succeeded at 2390 (offset 6 lines).
Hunk #8 succeeded at 4070 (offset 10 lines).
Hunk #9 succeeded at 4590 (offset 10 lines).
Hunk #10 succeeded at 5995 (offset 10 lines).
Hunk #11 succeeded at 7485 (offset 19 lines).
1 out of 11 hunks FAILED -- saving rejects to file module-webif.c.rej
patching file oscam-chk.c
Hunk #1 succeeded at 706 (offset 1 line).
Hunk #2 succeeded at 742 (offset 1 line).
[... 109 lines cut ...]
CC module-gbox-helper.c
CC module-gbox-sms.c
CC module-gbox-cards.c
CC module-gbox.c
CC module-ird-guess.c
CC module-monitor.c
CC module-newcamd.c
CC module-newcamd-des.c
CC module-pandora.c
CC module-ghttp.c
CC module-radegast.c
CC module-scam.c
CC module-serial.c
CC module-stat.c
CC module-webif-lib.c
CC module-webif-tpl.c
CC module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:5999:24: error: 'MNU_CFG_KEY' undeclared (first use in this function)
module-webif.c:5999:24: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/mipsel-unknown-linux-gnu/module-webif.o] Error 1
make: *** [all] Error 2

NO oscam-emu.patch OK
please fixed oscam-emu.patch

jahntheman, 2015-08-15:
thank you, fixed.

closed 2015-08-15 by jahntheman

commando2015, 2015-08-15:
thanks @jahntheman

commando2015, 2015-09-09:
10987 not ok

apply patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_064_long.h
patching file ffdecsa/parallel_128_sse2.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
Hunk #1 succeeded at 420 (offset -1 lines).
Hunk #2 succeeded at 834 (offset -1 lines).
Hunk #3 succeeded at 964 (offset -1 lines).
Hunk #4 succeeded at 1650 (offset 23 lines).
Hunk #5 succeeded at 2117 (offset 32 lines).
Hunk #6 succeeded at 2297 (offset 36 lines).
patching file Makefile
patching file module-dvbapi.c
Hunk #1 succeeded at 1221 (offset 12 lines).
Hunk #2 succeeded at 1248 (offset 12 lines).
Hunk #3 succeeded at 3958 with fuzz 2 (offset 133 lines).
Hunk #4 succeeded at 3996 (offset 145 lines).
Hunk #5 succeeded at 5357 with fuzz 2 (offset 166 lines).
patching file module-dvbapi.h
patching file module-emulator-dre2overcrypt.c
patching file module-emulator-dre2overcrypt.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-st20.c
patching file module-emulator-st20.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
Hunk #2 FAILED at 123.
Hunk #3 succeeded at 947 (offset 5 lines).
Hunk #4 succeeded at 1378 (offset 5 lines).
Hunk #5 succeeded at 1819 (offset 5 lines).
Hunk #6 succeeded at 2376 (offset 5 lines).
Hunk #7 succeeded at 2403 (offset 5 lines).
Hunk #8 succeeded at 4083 (offset 5 lines).
Hunk #9 succeeded at 4603 (offset 5 lines).
Hunk #10 succeeded at 6033 with fuzz 2 (offset 8 lines).
Hunk #11 succeeded at 7506 (offset 8 lines).
1 out of 11 hunks FAILED -- saving rejects to file module-webif.c.rej
patching file oscam-chk.c
patching file oscam-config-global.c
Hunk #1 succeeded at 803 (offset 1 line).
Hunk #2 succeeded at 1083 (offset 1 line).
patching file oscam-config-reader.c
patching file oscam-config.h
Hunk #2 succeeded at 61 (offset 5 lines).
patching file oscam-ecm.c
patching file oscam-emm.c
[... 96 lines cut ...]
CC	module-cw-cycle-check.c
CC	module-dvbapi-chancache.c
CC	module-dvbapi.c
CC	module-gbox-helper.c
CC	module-gbox-sms.c
CC	module-gbox-cards.c
CC	module-gbox.c
CC	module-ird-guess.c
CC	module-monitor.c
CC	module-newcamd.c
CC	module-newcamd-des.c
CC	module-pandora.c
CC	module-ghttp.c
CC	module-radegast.c
CC	module-scam.c
CC	module-serial.c
CC	module-stat.c
CC	module-webif-lib.c
CC	module-webif-tpl.c
CC	module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:6037:24: error: 'MNU_CFG_FSOFTCAMKEY' undeclared (first use in this function)
module-webif.c:6037:24: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/mipsel-unknown-linux-gnu/module-webif.o] Error 1
make: *** [all] Error 2