oscam-emu/oscam-emu #155: OSCam Changeset [10935] with oscam-emu.patch 0433392 #151

JanGruuthuse, 2015-08-01:
patching file config.h
Hunk #1 succeeded at 1 with fuzz 2.
patching file config.sh
Hunk #4 FAILED at 344.
1 out of 5 hunks FAILED -- saving rejects to file config.sh.rej

patching file module-webif.c
Hunk #2 FAILED at 120.
Hunk #3 succeeded at 939 (offset 2 lines).
Hunk #4 succeeded at 1365 (offset 2 lines).
Hunk #5 succeeded at 1806 (offset 2 lines).
Hunk #6 succeeded at 2363 (offset 6 lines).
Hunk #7 succeeded at 2390 (offset 6 lines).
Hunk #8 succeeded at 4070 (offset 10 lines).
Hunk #9 succeeded at 4590 (offset 10 lines).
Hunk #10 succeeded at 5995 (offset 10 lines).
Hunk #11 succeeded at 7485 (offset 19 lines).
1 out of 11 hunks FAILED -- saving rejects to file module-webif.c.rej

patching file webif/files/menu.html
Hunk #1 FAILED at 10.
Hunk #2 FAILED at 18.
2 out of 2 hunks FAILED -- saving rejects to file webif/files/menu.html.rej

CC	module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:5999:24: error: 'MNU_CFG_KEY' undeclared (first use in this function)
   { "SoftCam.Key",     MNU_CFG_KEY,       FTYPE_CONFIG },
                        ^
module-webif.c:5999:24: note: each undeclared identifier is reported only once for each function it appears in
CC	reader-common.c
make[1]: *** [build/mipsel-openpli40-linux-gnu/module-webif.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC	reader-bulcrypt.c
make: *** [all] Error 2

find details
- build.log
- config.sh.rej
- menu.html.rej
- module-webif.c.rej
@ https://gist.github.com/JanGruuthuse/0d4befad2b147a4c6aac

commando2015, 2015-08-02:
10918 OK support oscam-emu (oscam-svn = oscam-emu.patch)

OR

ERROR LOG
failed to build oscam-svn 10936-mipsoe20-webif-oscam-emu-patched

apply patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
Hunk #4 FAILED at 344.
1 out of 5 hunks FAILED -- saving rejects to file config.sh.rej
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
Hunk #1 succeeded at 421 (offset 17 lines).
Hunk #2 succeeded at 835 (offset 17 lines).
Hunk #3 succeeded at 965 (offset 17 lines).
Hunk #4 succeeded at 1627 (offset 27 lines).
Hunk #5 succeeded at 2085 (offset 39 lines).
Hunk #6 succeeded at 2258 (offset 42 lines).
patching file Makefile
patching file module-dvbapi.c
Hunk #1 succeeded at 1209 (offset 56 lines).
Hunk #2 succeeded at 1236 (offset 56 lines).
Hunk #3 succeeded at 3825 (offset 58 lines).
Hunk #4 succeeded at 3851 (offset 58 lines).
Hunk #5 succeeded at 5191 (offset 104 lines).
patching file module-dvbapi.h
Hunk #1 succeeded at 98 (offset 1 line).
patching file module-emulator-dre2overcrypt.c
patching file module-emulator-dre2overcrypt.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-st20.c
patching file module-emulator-st20.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
Hunk #2 FAILED at 120.
Hunk #3 succeeded at 939 (offset 2 lines).
Hunk #4 succeeded at 1365 (offset 2 lines).
Hunk #5 succeeded at 1806 (offset 2 lines).
Hunk #6 succeeded at 2363 (offset 6 lines).
Hunk #7 succeeded at 2390 (offset 6 lines).
Hunk #8 succeeded at 4070 (offset 10 lines).
Hunk #9 succeeded at 4590 (offset 10 lines).
Hunk #10 succeeded at 5995 (offset 10 lines).
Hunk #11 succeeded at 7485 (offset 19 lines).
1 out of 11 hunks FAILED -- saving rejects to file module-webif.c.rej
patching file oscam-chk.c
Hunk #1 succeeded at 706 (offset 1 line).
Hunk #2 succeeded at 742 (offset 1 line).
[... 106 lines cut ...]
CC module-cw-cycle-check.c
CC module-dvbapi-chancache.c
CC module-dvbapi.c
CC module-gbox-helper.c
CC module-gbox-sms.c
CC module-gbox-cards.c
CC module-gbox.c
CC module-ird-guess.c
CC module-monitor.c
CC module-newcamd.c
CC module-newcamd-des.c
CC module-pandora.c
CC module-ghttp.c
CC module-radegast.c
CC module-scam.c
CC module-serial.c
CC module-stat.c
CC module-webif-lib.c
CC module-webif-tpl.c
CC module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:5999:24: error: 'MNU_CFG_KEY' undeclared (first use in this function)
module-webif.c:5999:24: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/mipsel-unknown-linux-gnu/module-webif.o] Error 1
make: *** [all] Error 2

jahntheman, 2015-08-15:
thank you, fixed.

closed 2015-08-15 by jahntheman

commando2015, 2015-08-15:
thanks @jahntheman