oscam-emu/oscam-emu #156: patch not apply corectly in svn10937

echosmart792, 2015-08-03:
patch -p0 < ../oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
Hunk #4 FAILED at 344.
1 out of 5 hunks FAILED -- saving rejects to file config.sh.rej
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
Hunk #1 succeeded at 421 (offset 17 lines).
Hunk #2 succeeded at 835 (offset 17 lines).
Hunk #3 succeeded at 965 (offset 17 lines).
Hunk #4 succeeded at 1627 (offset 27 lines).
Hunk #5 succeeded at 2085 (offset 39 lines).
Hunk #6 succeeded at 2258 (offset 42 lines).
patching file Makefile
patching file module-dvbapi.c
Hunk #1 succeeded at 1209 (offset 56 lines).
Hunk #2 succeeded at 1236 (offset 56 lines).
Hunk #3 succeeded at 3823 (offset 56 lines).
Hunk #4 succeeded at 3849 (offset 56 lines).
Hunk #5 succeeded at 5189 (offset 102 lines).
patching file module-dvbapi.h
Hunk #1 succeeded at 98 (offset 1 line).
patching file module-emulator-dre2overcrypt.c
patching file module-emulator-dre2overcrypt.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-st20.c
patching file module-emulator-st20.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
Hunk #2 FAILED at 120.
Hunk #3 succeeded at 939 (offset 2 lines).
Hunk #4 succeeded at 1365 (offset 2 lines).
Hunk #5 succeeded at 1806 (offset 2 lines).
Hunk #6 succeeded at 2363 (offset 6 lines).
Hunk #7 succeeded at 2390 (offset 6 lines).
Hunk #8 succeeded at 4070 (offset 10 lines).
Hunk #9 succeeded at 4590 (offset 10 lines).
Hunk #10 succeeded at 5995 (offset 10 lines).
Hunk #11 succeeded at 7485 (offset 19 lines).
1 out of 11 hunks FAILED -- saving rejects to file module-webif.c.rej
patching file oscam-chk.c
Hunk #1 succeeded at 706 (offset 1 line).
Hunk #2 succeeded at 742 (offset 1 line).
Hunk #3 succeeded at 975 (offset 1 line).
patching file oscam-config-global.c
Hunk #1 succeeded at 802 (offset 1 line).
Hunk #2 succeeded at 1081 (offset 1 line).
patching file oscam-config-reader.c
Hunk #2 succeeded at 446 (offset 44 lines).
Hunk #3 succeeded at 754 (offset 44 lines).
Hunk #4 succeeded at 889 (offset 46 lines).
patching file oscam-config.h
Hunk #2 succeeded at 56 (offset 1 line).
patching file oscam-emm.c
patching file oscam-string.c
Hunk #1 succeeded at 494 (offset 3 lines).
patching file oscam-string.h
patching file oscam.c
Hunk #3 succeeded at 1585 (offset 1 line).
Hunk #4 succeeded at 1759 (offset 1 line).
Hunk #5 succeeded at 1841 (offset 2 lines).
patching file reader-common.c
patching file readers.h
patching file webif/config/menu.html
patching file webif/config/menu_streamrelay.html
patching file webif/config/streamrelay.html
patching file webif/files/menu.html
Hunk #1 FAILED at 10.
Hunk #2 FAILED at 18.
2 out of 2 hunks FAILED -- saving rejects to file webif/files/menu.html.rej
patching file webif/files/menu_key.html
patching file webif/pages_index.txt
Hunk #3 succeeded at 179 (offset 2 lines).
patching file webif/readerconfig/readerconfig_emubit.html

echosmart792, 2015-08-03:
if i want to compile patched version this is buildlog:

+-------------------------------------------------------------------------------
| OSCam ver: 1.20-unstable_svn rev: 10937 target: mipsel-unknown-linux-gnu
| Tools:
|  CROSS    = /xxxxxxxxxxxxxxxx/oscam_emu/toolchains/mipsoe20/bin/mipsel-unknown-linux-gnu-
|  CC       = /xxxxxxxxxxxxxxx/oscam_emu/toolchains/mipsoe20/bin/mipsel-unknown-linux-gnu-gcc
| Settings:
|  CONF_DIR = /usr/keys/oscam
|  CC_OPTS  = -Os -ggdb -pipe -ffunction-sections -fdata-sections
|  CC_WARN  = -w
|  CFLAGS   =
|  LDFLAGS  = -Wl,--gc-sections -Wl,-rpath,/lib -Wl,-rpath,/usr/local/lib -Wl,-rpath,/usr/lib -L/home/toshiba/oscam_emu/toolchains/mipsoe20/mipsel-unknown-linux-gnu/sysroot/usr/lib
|  LIBS     = -lpthread -ldl -lrt
|  UseFlags =
| Config:
|  Addons   : WEBIF WEBIF_LIVELOG HAVE_DVBAPI READ_SDT_CHARSETS CS_ANTICASC MODULE_MONITOR WITH_LB CS_CACHEEX CW_CYCLE_CHECK CLOCKFIX WITH_EMU
|  Protocols: NEWCAMD CCCAM CCCSHARE CONSTCW
|  Readers  : NAGRA CONAX CRYPTOWORKS VIACCESS
|  CardRdrs : INTERNAL
|  Compiler : mipsel-unknown-linux-gnu-gcc (crosstool-NG 1.15.2) 4.6.3
|  Config   : build/mipsel-unknown-linux-gnu/config.mak
|  Binary   : ../archiv/oscam-svn10937-mipsoe20-webif
+-------------------------------------------------------------------------------
HOSTCC   webif/pages_gen
GEN   webif/pages.c
GEN   Compressed 269856 template bytes into 102903 bytes. 166953 saved bytes (61.87%).
CC   cscrypt/bn_add.c
CC   cscrypt/aes.c
CC   cscrypt/bn_ctx.c
CC   cscrypt/bn_asm.c
CC   cscrypt/bn_div.c
CC   cscrypt/bn_exp.c
CC   cscrypt/bn_lib.c
CC   cscrypt/bn_mul.c
CC   cscrypt/bn_print.c
CC   cscrypt/bn_shift.c
CC   cscrypt/bn_sqr.c
CC   cscrypt/bn_word.c
CC   cscrypt/mem.c
CC   cscrypt/i_cbc.c
CC   cscrypt/des.c
CC   cscrypt/i_ecb.c
CC   cscrypt/i_skey.c
CC   cscrypt/md5.c
CC   cscrypt/rc6.c
CC   cscrypt/sha1.c
CC   csctapi/atr.c
CC   csctapi/icc_async.c
CC   csctapi/io_serial.c
CC   csctapi/protocol_t0.c
CC   csctapi/protocol_t1.c
CC   csctapi/ifd_sci.c
CC   minilzo/minilzo.c
CC   module-cacheex.c
CC   module-anticasc.c
CC   module-camd35-cacheex.c
CC   module-cccam-cacheex.c
CC   module-cccam.c
CC   module-cccshare.c
CC   module-constcw.c
CC   module-emulator.c
CC   module-emulator-osemu.c
CC   module-emulator-stream.c
CC   module-emulator-st20.c
CC   module-emulator-dre2overcrypt.c
CC   ffdecsa/ffdecsa.c
CC   module-csp.c
CC   module-cw-cycle-check.c
CC   module-dvbapi-chancache.c
CC   module-dvbapi.c
CC   module-monitor.c
CC   module-newcamd.c
CC   module-newcamd-des.c
CC   module-stat.c
CC   module-webif-lib.c
CC   module-webif-tpl.c
CC   module-webif.c
CC   webif/pages.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:5999:24: error: 'MNU_CFG_KEY' undeclared (first use in this function)
module-webif.c:5999:24: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [build/mipsel-unknown-linux-gnu/module-webif.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC   reader-common.c
make: *** [all] Error 2

jahntheman, 2015-08-15:
thank you, fixed.

closed 2015-08-15 by jahntheman