oscam-emu/oscam-emu #161: Cannot descramble BISS without CA descriptor

glamourservices, 2015-08-16:
1. In some BISS encrypted channels there is no CA descriptor, thus the e2 receiver perceives them as FTA. Notable examples are ERT from 3E and BBC from 24.5W. Is it possible to find a solution to detect and descramble these biss feeds without checking for their ca descriptor?
2. In Oscam emu all emulated caids are shown as cards, whereas they should be shown as EMU, is this legit or is it the only way to implement an emulation function in the oscam code?

JanGruuthuse, 2015-08-17:
Intelsat 907 (27.5°W) BBC-SATBACK instead of 24.5W.
Legit? This is for educational purposes only!

jahntheman, 2015-10-03:
1. Please check oscam dvbapi mapping config, this should fix it maybe. If not please go to official oscam forum to discuss it.

2. What do you mean with "should be shown as EMU? You can give the reader any name you like and you can always check its reader protocol ("emu").

closed 2015-10-03 by jahntheman