oscam-emu/oscam-emu #162: HEX Number entry in SoftCam.Key file

FCS001FCS, 2015-08-17:
The README text for the POWERVU section may need to be clarified.
I am informed (rightly or wrongly?) by another site that when you use the Service Identification (srvid) for PowerVu services (maybe all services??) in the SoftCam.Key file the first digit of the srvid Hex number needs to be moved to the end of the number. For example, I am using OpenPLi 4 and when I get the srvid from the information menu for a PowerVu channel it is for example Hex 1234 but in the SoftCam.Key file it needs to be entered as 2341. Do not know why but maybe you can explain and is this for all encryption systems not just Powervu? Maybe a note should be added to the Readme file? Note that the stream relay setup entry for the svrid entered in Dreamboxedit is entered as Decimal not Hex.

JanGruuthuse, 2015-08-17:
You can discuss: http://www.streamboard.tv/wbb2/thread.php?threadid=39010
The readme in source tells:

PowerVU config
==============

Keys in SoftCam.Key:
	P <srvid> <nb> <key>
	example:
	P 1234 01 11223344556677

	For channels with same srvid:
	Channel #1: P  1234 01 11223344556677
	Channel #2: P 11234 01 11223344556677
	Channel #3: P 21234 01 11223344556677
	
	AU keys:
	P <srvid> <UA> <key>
	example:
	P 1234 11223344 11223344556677

For PowerVU the Stream Relay must be used:

IMPORTANT:
	The stream relay needs a fast CPU. Not every STB is fast enough.
	If you cannot fix stuttering with the following guide, likely your CPU is 
	too slow.

	For best performance, the "Include ECM in http streams" setting must be enabled.
	If your image supports it (openpli, openatv), you can find it here:
	OpenWebif -> Settings -> Customize
	
	If the setting is not available, you may need to adjust the "ECM fix delay"
	setting in oscam webif (Config -> Stream Relay).
	Increase/Lower it until there is no stuttering.

	When using VU+ devices, make sure to use the latest drivers (June 2015), if 
	you encounter any problems.

Channel Config:
1. Go to the OpenWebif and click on the "Stream" button for the channel you want to add.
2. Download and save the .m3u file.
3. Open the .m3u file with a text edior and get the stream link. It looks like this:
	http://stb:8001/1:0:19:0123:3EF:1:C00000:0:0:0:
4. Replace the port (8001) with the relay port (default: 17999) and the host name ("stb") with 127.0.0.1
	http://127.0.0.1:17999/1:0:19:0123:3EF:1:C00000:0:0:0:
5. Add the new stream link to your channel list (for example with DreamBoxEdit)

IMPORTANT:
	Do NOT use the correct Transponder ID/Network ID/Namespace for the stream link.
	Use Transponder ID: "11", Network ID: "22" and Namespace: "33".
	Set Service Type to "1", and enter the correct Service ID for the channel.

FCS001FCS, 2015-08-17:
OK will move topic to Streamboard.

closed 2015-08-17 by FCS001FCS