oscam-emu/oscam-emu #163: module-emulator-dre2overcrypt.c Compile error.

heugabel, 2015-08-17:
message: "sse2 instruction set not enabled" 
for including  <emmintrin.h> in ffdecsa/parallel_128_sse2.h

| OSCam ver: 1.20-unstable_svn rev: 10943 target: i486-linux-gnu-libusb

gianni8127, 2015-08-19:
/usr/lib/gcc/i686-linux-gnu/4.4.5/include/emmintrin.h:32: error: #error "SSE2 instruction set not enabled"
make[1]: *** [build/i686-linux-gnu/ffdecsa/ffdecsa.o] Errore 1
make: *** [all] Errore 2

gianni8127, 2015-08-23:
please add fix for compiled

joeusercz, 2015-08-23:
Try adding option to make command line:
EXTRA_FLAGS="-DPARALLEL_MODE=PARALLEL_32_INT"

gianni8127, 2015-08-26:
not change

gianni8127, 2015-08-26:
please divide the functions emulatore and streamrelay

heugabel, 2015-08-31:
go to /usr/lib/gcc/i486-linux-gnu/X.X/include

open emmintrin.h, mmintrin.h and xmmintrin.h

search for # error "SSE2 instruction set not enabled"

change to /*# error "SSE2 instruction set not enabled"*/

This tryfix works for me.

gianni8127, 2015-09-07:
The new patch compiles without errors :+1: 
git not updated regularly. Look for latest patch here: http://www.streamboard.tv/wbb2/thread.php?threadid=39010&threadview=1&hilight=&hilightuser=0&page=84#post551427

jahntheman, 2015-10-03:
fixed

closed 2015-10-03 by jahntheman