oscam-emu/oscam-emu #165: Alternative biss key format

nautilus7, 2015-08-30:
Is it possible to implement an alternative format to enter biss keys? Currently the supported format is

F [SID][videoPID] 00 KEY
F [SID][videoPID] 01 KEY

e.g. F 22221FFF 00 0102030405060708
       F 22221FFF 01 0102030405060708

Problem is that SID is not enough and there can easily exist more than one services with the same SID, so only one of them will work. I would like to suggest an alternative format for entering biss keys, like this

F [ONID][SID][videoPID] 00 KEY

e.g. F 006400380808 00 9DE4C142AC513C39

This is the same format supported by oscam_ymod.

jahntheman, 2015-10-03:
If you need advanced BISS key support, please use a constcw reader for it. Check official oscam wiki and forum for details.

The "F [SID][EcmPID] 00 KEY" format is the only one possible for oscam-emu/OSemu, because it is the only one that can be shared.

closed 2015-10-03 by jahntheman

nautilus7, 2015-10-03:
What do you mean by "it's the only one that can be shared"??? Please explain!

jahntheman, 2015-10-03:
protocols like camd35, cccam, newcamd, etc only send SID and sometimes ecmpid if you are lucky. ONID and videoPID cannot be transmitted by any of these protocols.
Thus ONID and videoPID are unknown when the request arrives.

nautilus7, 2015-10-03:
I see now. Thanks.