oscam-emu/oscam-emu #166: Biss key not working at all

taktiker12, 2015-08-31:
hi ,
trying this reader 

[reader]
label = oscam_emu
protocol = emu
device = Emulator

caid =  0D00,0D02,090F,0500,1801,0604,2600,FFFF,0E00,4AE1

detect = cd

ident = 0D00:000000,000004,000010,000014,000020,0000C0,0000C4,0000C8,0000CC;0D02:000000,00008C,0000A0,0000A4,0000A8;090F:000000;0500:000000,030B00,023800,021110,007400,007800;1801:000000,007301,001101;0604:000000;2600:000000;FFFF:000000;0E00:000000;4AE1:000000

group = 1
emmcache = 1,2,2,1
emu_auproviders = 0500:030B00;0604:010200;0E00:000000;4AE1:000000

and  SoftCam.Key . 
The Channel eleven_sports on Hotbird does not open and many others  too . i use vdr with dvbapi and latest oscam patched . 
sometimes it opens for 20 Seconds and stopped decoding with error no matching reader. 

Thnx in Advance

Yuri666, 2015-09-01:
From dvbapi readme:

Disable CW expiration check:
----------------------------
The plugin is automatically checking if the key obtained from OSCam is still
valid and doesn't expired before using it for decrypting. This way if OSCam
cannot provide a key for a channel, VDR have a chance to switch to other CAM
which is able to do so. In some circumstances (eg. when you're using Constant
CW in OSCam) you may need to disable this check, starting VDR like this:
  vdr -P"dvbapi -d"

taktiker12, 2015-09-01:
Thank You Youri
I will try this solution  and give you feedback .

taktiker12, 2015-09-01:
constant.cw + oscam+ vdr with dvbapi -d works after i added the PMT PID to constant.cw. 
thank you

closed 2015-10-03 by jahntheman