oscam-emu/oscam-emu #170: OSCam ver: 1.20-unstable_svn rev: 10991 build error

Matthias199, 2015-09-14:
Hey if I try to compile this version with the patch there is in compile error

CC      module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:6037: error: 'MNU_CFG_FSOFTCAMKEY' undeclared (first use in this function)
module-webif.c:6037: error: (Each undeclared identifier is reported only once
module-webif.c:6037: error: for each function it appears in.)
make[1]: *** [build/arm-none-linux-gnueabi-libusb/module-webif.o] Error 1
make: *** [all] Error 2

without the path is everything working fine any idea?

QRS01, 2015-09-16:
The patch mentioned by joeusercz some issues ago, is still OK... 
Look for latest patch here: http://www.streamboard.tv/wbb2/thread.php?threadid=39010&threadview=1&hilight=&hilightuser=0&page=84#post551427

Matthias199, 2015-09-16:
tanks but why there is not an master upgrade ?

jahntheman, 2015-10-03:
fixed

closed 2015-10-03 by jahntheman