oscam-emu/oscam-emu #172: PowerVu: problem selecting the correct s-box for each ecm

Purrfect4711, 2015-10-01:
Hi!

"For EMM decryption the sbox set can be chosen during the personalisation of the ISE and can't be changed during the runtime.
For ECM decryption the sbox set can be chosen by a bit in the ECM. The same byte in nano 20 that selects the even/odd key can control with a set bit 3 (mask 0x08) that the alternate sboxes are used." In a forum I also found "first try with sbox mode = 0 (primary tables), if it works, great, otherwise test the 0xA0 mode (secondary tables)."

Unfortunately oscam-emu doesn't support this, yet. Is it possible to add this feature?

Thanks and regards,
Purrfect

jahntheman, 2015-10-03:
Thank you. 
Has been fixed.

closed 2015-10-03 by jahntheman

budiarno, 2016-01-08:
Hi, it seems this problem still persist. 

First screenshot is the working one on OSCam-Emu:
[image]

Now, the channel has changed:
[image]

We only get blank picture on these channels (for example on TCM at Asiasat7 and Astro on Measat 3). Wish there a solution for this problem.