oscam-emu/oscam-emu #179: PowerVU AU doesn't work anymore

Purrfect4711, 2015-10-27:
I've been using oscam-emu for several weeks now and I noticed that PowerVU AU is no longer working. Updating from r10991 to r11034 didn't help, too. I changed the loglevel to 64 and see the following entry in the logfile when I try to write an EMM manually:

2015/10/26 10:11:02 3A10C2FB h (reader) emulator [emu] EMM: get_emm_type() returns error
2015/10/26 10:11:02 3A10C2FB h (webif) emm is being sent to reader emulator.
2015/10/26 10:11:02 43D3346D r (emmcache) added emm to cache:
2015/10/26 10:11:02 43D3346D r (emmcache) 13 2B 19 F9 EF 4A 20 6C 6E B0 3F 58 AA 55 6B 9E
2015/10/26 10:11:02 43D3346D r (emmcache) added emmstat (reader:emulator, count:0)
2015/10/26 10:11:02 43D3346D r (emmcache) 13 2B 19 F9 EF 4A 20 6C 6E B0 3F 58 AA 55 6B 9E
2015/10/26 10:11:02 43D3346D r [Emu] EMM failed: EMM not supported

The EMM is added to the cache but not really supported. About 3 weeks ago it used to work as this appeared in SoftCam.Key:  P 0020 00 xxxxx  ; added by OSEmu Fri Oct  2 11:22:34 2015. I presume the provider altered the EMMs in a way that oscam-emu cannot handle them?

Purrfect4711, 2015-10-27:
Update: the provider I'm receiving didn't send any EMMs for days but now they started again. I see the following entry in my logfile:

2015/10/27 10:07:41 70BBE540 c      (emm) emm:
2015/10/27 10:07:41 70BBE540 c      (emm)   82 30 9B 10 99 01 0E 00 00 00 06 8F 00 60 5A CD
2015/10/27 10:07:41 70BBE540 c      (emm)   00 00 03 80 88 8D A1 B8 6F A9 AF 8C E7 95 48 57
2015/10/27 10:07:41 70BBE540 c      (emm)   C3 BB FC 83 E6 32 38 75 14 68 76 29 6C 2B 80 88
2015/10/27 10:07:41 70BBE540 c      (emm)   8C C1 CD 92 14 F6 3E F3 B2 61 D3 B8 AB F8 52 A5
2015/10/27 10:07:41 70BBE540 c      (emm)   A5 21 1B 25 05 AA 49 A4 B3 80 88 8F 80 BE 69 38
2015/10/27 10:07:41 70BBE540 c      (emm)   C1 BE 41 88 14 E4 EF 60 2B 83 B1 0E 4C 51 E8 9D
2015/10/27 10:07:41 70BBE540 c      (emm)   51 45 9A 8C 80 88 8B 02 EE 51 50 32 89 FA 7E 1E
2015/10/27 10:07:41 70BBE540 c      (emm)   33 02 57 CE AE 2A B0 DE E1 15 99 A0 AB 55 12 80
2015/10/27 10:07:41 70BBE540 c      (emm)   88 8E 78 45 FC AE 09 F5 9E C4 C5 0F 49 BF 50 A2
2015/10/27 10:07:41 70BBE540 c      (emm)   45 8A D9 1D 77 24 14 1E C0 56 64 BE E6 A4
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emmtype unique. Reader serial ################.
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emm UA/SA: ################.
2015/10/27 10:07:41 70BBE540 c (emmcache) added emmstat (reader:emulator, count:0)
2015/10/27 10:07:41 70BBE540 c (emmcache)   50 B4 B9 7E FA 19 9C 5B 4C F5 01 E5 37 91 40 FB
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emm count 0 rewrite 1
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emm is being sent to reader
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] Successfully added EMM to /var/log/oscam/emulator_unique_emm.log
2015/10/27 10:07:41 43D3346D r (emmcache) added emm to cache:
2015/10/27 10:07:41 43D3346D r (emmcache)   50 B4 B9 7E FA 19 9C 5B 4C F5 01 E5 37 91 40 FB
2015/10/27 10:07:41 43D3346D r (emmcache) found emmstat match (reader:emulator, count:0)
2015/10/27 10:07:41 43D3346D r (emmcache)   50 B4 B9 7E FA 19 9C 5B 4C F5 01 E5 37 91 40 FB
2015/10/27 10:07:41 43D3346D r   (reader) emulator [emu] mumudvb emmtype=unique, len=155 (hex: 0x9B), cnt=1: written (1 ms)
2015/10/27 10:07:41 70BBE540 c      (emm) emm:
2015/10/27 10:07:41 70BBE540 c      (emm)   82 30 65 10 63 01 0E 00 00 00 06 59 00 60 5A CD
2015/10/27 10:07:41 70BBE540 c      (emm)   00 00 03 00 0C 08 80 4C 00 00 FF FF 00 00 00 00
2015/10/27 10:07:41 70BBE540 c      (emm)   00 00 00 00 00 00 00 00 00 00 00 A6 84 0D 00 0C
2015/10/27 10:07:41 70BBE540 c      (emm)   10 11 06 08 BA DC 11 06 08 BA DC 11 06 08 BA DC
2015/10/27 10:07:41 70BBE540 c      (emm)   11 06 08 BA DC 00 59 1E AD 00 0C 12 F7 E0 FF C0
2015/10/27 10:07:41 70BBE540 c      (emm)   00 00 00 00 00 00 F0 80 88 90 98 00 00 00 00 00
2015/10/27 10:07:41 70BBE540 c      (emm)   00 60 5F 5C B6 E8 9C 42
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emmtype unique. Reader serial ################.
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emm UA/SA: ################.
2015/10/27 10:07:41 70BBE540 c (emmcache) found emmcache match
2015/10/27 10:07:41 70BBE540 c (emmcache)   22 99 86 85 C0 FE C0 27 41 07 54 84 1A CF AF 7E
2015/10/27 10:07:41 70BBE540 c (emmcache) found emmstat match (reader:emulator, count:1)
2015/10/27 10:07:41 70BBE540 c (emmcache)   22 99 86 85 C0 FE C0 27 41 07 54 84 1A CF AF 7E
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] EMM: emm count 1 rewrite 1
2015/10/27 10:07:41 70BBE540 c   (reader) emulator [emu] Successfully added EMM to /var/log/oscam/emulator_unique_emm.log

Looks like it's working if the EMMs are coming in "live" but the above issue still happens if I try to manually write an EMM.

jahntheman, 2015-10-27:
"[Emu] EMM failed: EMM not supported" means the EMM is too short.
Normally this only happens for EMMs that do not contain keys (these emms are not encrypted). But it is certainly possible that there have been some changes to the encrypted EMMs. Can you post an EMM that gives the "EMM not supported" message? And be careful when posting EMMs ,they contain the UA (you might want to edit your posts).

jahntheman, 2015-10-27:
UPDATE
Manual EMM writing was broken and has been fixed with v723.
Always enter the correct CAID when manual writing EMMs.

Purrfect4711, 2015-10-31:
It's working now, thank you very much. :-)

closed 2015-10-31 by jahntheman