oscam-emu/oscam-emu #189: on svn 11089 with emu, IDX Xtra Thor 1*W pictures slide show whithout sound.

echosmart792, 2015-11-05:
Hi with new svn 11089 and new patch from this page, IDX Xtra working only picture slide show whithout sound. In earlier versions (11070 or earliers) this channel working continuosly sound and picture. 

i don't try other channels with svn 11089.

jahntheman, 2015-11-05:
can you please post the pmt or capmt of this channel?

echosmart792, 2015-11-07:
How to post this ?

echosmart792, 2015-11-07:
From dvbapi ?

PMT Mode = 0

echosmart792, 2015-11-07:
This is from LiveLOG:

2015/11/07 08:17:37 00000000   [Emu] stream client disconnected
2015/11/07 08:17:38 00000000   [Emu] stream client connected
2015/11/07 08:17:38 00000000   [Emu] stream client request /1:0:1:BB8:2:5E:E08B00F:0:0:0:
2015/11/07 08:17:38 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:17:39 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:17:44 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:17:45 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:17:46 00000000 [Emu] stream client disconnected
2015/11/07 08:17:46 00000000 [Emu] stream client connected
2015/11/07 08:17:46 00000000 [Emu] stream client request /1:0:1:BB9:2:5E:E08B00F:0:0:0:
2015/11/07 08:17:48 424D185D r [Emu] ECM failed: Corrupt data

All channel slow slide show whithout sound.

echosmart792, 2015-11-07:
This is Discovery HD Showcase from Thor

2015/11/07 08:33:49 424D185D r [Emu] ECM failed: Corrupt data
2015/11/07 08:33:53 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:33:54 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:33:57 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:33:59 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:00 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:06 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:18 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:25 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:25 424D185D r [Emu] ECM failed: ECM not supported
2015/11/07 08:34:27 424D185D r [Emu] ECM failed: Corrupt data
2015/11/07 08:34:28 424D185D r [Emu] ECM failed: ECM not supported

But this channel workin OK !
Rarely freezeng.

echosmart792, 2015-11-07:
This is from livelog:

2015/11/07 08:45:53 00000000   [Emu] stream client disconnected
2015/11/07 08:45:53 00000000   [Emu] stream client connected
2015/11/07 08:45:53 00000000   [Emu] stream client request /1:0:1:9C4:2:5E:E08B00F:0:0:0:
2015/11/07 08:46:11 00000000   [Emu] stream client disconnected
2015/11/07 08:46:11 00000000   [Emu] stream client connected
2015/11/07 08:46:11 00000000   [Emu] stream client request /1:0:1:A2D:2:5E:E08B00F:0:0:0:
2015/11/07 08:46:17 00000000   [Emu] stream client disconnected
2015/11/07 08:46:17 00000000   [Emu] stream client connected
2015/11/07 08:46:17 00000000   [Emu] stream client request /1:0:1:A28:2:5E:E08B00F:0:0:0:
2015/11/07 08:46:20 00000000   [Emu] stream client disconnected
2015/11/07 08:46:21 00000000   [Emu] stream client connected
2015/11/07 08:46:21 00000000   [Emu] stream client request /1:0:1:9C9:2:5E:E08B00F:0:0:0:

But channel working only slide show.

echosmart792, 2015-11-07:
I stop all other from oscam for LiveLog acuratecy. The previos message it's accourate LiveLog, only Emu reader, and only DVBAPI user.

Channels runnung slide show picture whitout sound.
Strange, Discovery HD Showcase runnung, but livelog it's only this

2015/11/07 08:51:44 00000000   [Emu] stream client disconnected
2015/11/07 08:51:45 00000000   [Emu] stream client connected
2015/11/07 08:51:45 00000000   [Emu] stream client request /1:0:1:9CA:2:5E:E08B00F:0:0:0:

nothing else...

jahntheman, 2016-06-22:
some updates have been added.

closed 2016-06-22 by jahntheman