oscam-emu/oscam-emu #191: Oscam 11086 patched .. Why not  for VU+

Aliraza63, 2015-11-12:
Oscam 11086 patched work for Power VU keys in DM Why not it works in Vu Receivers 
I have tested in SOLO2 with Open black hole , Black hole  , Opena Atv , Eurosat Image to . it opens biss channels but when I swtich to any Power VU Oscam shows dcw but all 0000 and doesnt clear the picture . As I know both DM and VU have mips processor then why it doesnt work in VU +?
Kindly share it

joeusercz, 2015-11-13:
The processor is the same, but other chips are not. "IF" the vu supports hardware des, then there has to be drivers (and a known interface) in order for it to work.

loxel, 2015-11-14:
Is there a chance to PowerVu work on Vu+ as now runs on Dreambox ?

FCS001FCS, 2015-11-15:
Is there a VU+ expert that can confirm or reject that VU+ is capable of the same function (supports hardware des) as is in the Dreamboxes? If it supports hardware des, what needs to be done to get it working on Vu+ boxes?

Aliraza63, 2015-11-15:
When Peoples will be able to watch Power VU on VU +. its the question which answer is still pending

arcan62, 2015-12-12:
I checked powervu channels on vu+zero and its working the following sats which I received are working
66E,68.5E,76E but on asiasat 105.5E channels are not working.