oscam-emu/oscam-emu #193: powervu 2./3. audio problem hd channels

jamevu, 2015-11-26:
(dvbapi) PMT6: Trying connect to enigma CA PMT listen socket...
2015/11/26 23:51:37 69761DAD c   (dvbapi) PMT6 CA PMT Server connected on fd 12!
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 0E00 ECM_PID: 1F76 PROVID: 000000 
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 stream Videostream (MPEG-4)(type: 1b pid: 09c4 length: 0)
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 stream Data-/Audiostream (Subtitles/VBI and AC-3)(type: 06 pid: 09c5 length: 0)
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 stream Data-/Audiostream (Subtitles/VBI and AC-3)(type: 06 pid: 09c9 length: 0)
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 found 1 ECMpids and 3 STREAMpids in caPMT
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 serving srvid 09C8 (0E00@000000:09C8 unknown) on adapter 0000 camask 0001 index 0000 pmtpid 0A22
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 found channel in cache and matching prio -> start descrambling ecmpid 0 
2015/11/26 23:51:55 69761DAD c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1F76 ANY CHID PMTPID 0A22 VPID 09C4
2015/11/26 23:51:55 69761DAD c      (ecm) user1 (0E00@000000/0000/09C8/40:8218E4308B8DE589D7E8900074883A48): found (4 ms) by emulator
2015/11/26 23:51:56 69761DAD c      (ecm) user1 (0E00@000000/0000/09C8/40:3DE81F750465FE41FAA93FD7420952A1): found (7 ms) by emulator
2015/11/26 23:51:56 69761DAD c   (dvbapi) Demuxer 0 stream Videostream (MPEG-4)(type: 1b pid: 09c4 length: 23)
2015/11/26 23:51:56 69761DAD c   (dvbapi) Demuxer 0 stream Data-/Audiostream (Subtitles/VBI and AC-3)(type: 06 pid: 09c5 length: 10)
2015/11/26 23:51:56 69761DAD c   (dvbapi) Demuxer 0 stream Data-/Audiostream (Subtitles/VBI and AC-3)(type: 06 pid: 09c9 length: 10)
2015/11/26 23:51:56 69761DAD c   (dvbapi) Demuxer 0 found 1 ECMpids and 3 STREAMpids in PMT
2015/11/26 23:51:56 69761DAD c   (dvbapi) Demuxer 0 serving srvid 09C8 (0E00@000000:09C8 unknown) on adapter 0000 camask 0001 index 0000 pmtpid 0A22
2015/11/26 23:51:56 69761DAD c      (ecm) user1 (0E00@000000/0000/09C8/40:965513B84ABC1C4E73106DDEDDC8EA65): found (9 ms) by emulator
2015/11/26 23:51:57 69761DAD c      (ecm) user1 (0E00@000000/0000/09C8/40:60D652095FC6B56B82F7DB7511DE6563): found (5 ms) by emulator
2015/11/26 23:51:57 69761DAD c      (ecm) user1 (0E00@000000/0000/09C8/40:6743673867E10C0B5E49234E642F67BA): found (4 ms) by emulator

budiarno, 2015-11-27:
Use 128 Log mode for digging more on the error messages.

jamevu, 2015-11-27:
Here it is 128 debug log
[oscam.log.txt](https://github.com/oscam-emu/oscam-emu/files/46164/oscam.log.txt)

jamevu, 2015-11-27:
[image]

budiarno, 2015-11-28:
Did I miss "Audio PID" from your Service Info screenshot?

jamevu, 2015-12-01:
"Ses PID"  is turkish  same as "Audio PID"

joeusercz, 2016-01-01:
If you are using stream relay, you can use the new patch posted on streamboard.

littlesat, 2016-04-16:
I found a patch version that solves the audio AC3 issue... (e.g. discovery science HD 1W)
Below the diff..

3c3
< --- CMakeLists.txt	(revision 11212)
---
> --- CMakeLists.txt	(revision 11086)
96,146d95
< Index: Makefile
< ===================================================================
< --- Makefile	(revision 11212)
< +++ Makefile	(working copy)
< @@ -65,6 +65,13 @@
<  
<  LDFLAGS = -Wl,--gc-sections
<  
< +TARGETHELP := $(shell $(CC) --target-help)
< +ifneq (,$(findstring sse2,$(TARGETHELP)))
< +override CFLAGS += -fexpensive-optimizations -mmmx -msse -msse2 -msse3
< +else
< +override CFLAGS += -fexpensive-optimizations
< +endif
< +
<  # The linker for powerpc have bug that prevents --gc-sections from working
<  # Check for the linker version and if it matches disable --gc-sections
<  # For more information about the bug see:
< @@ -267,6 +274,23 @@
<  SRC-$(CONFIG_MODULE_CCCAM) += module-cccam.c
<  SRC-$(CONFIG_MODULE_CCCSHARE) += module-cccshare.c
<  SRC-$(CONFIG_MODULE_CONSTCW) += module-constcw.c
< +SRC-$(CONFIG_WITH_EMU) += module-emulator.c
< +SRC-$(CONFIG_WITH_EMU) += module-emulator-osemu.c
< +SRC-$(CONFIG_WITH_EMU) += module-emulator-stream.c
< +SRC-$(CONFIG_WITH_EMU) += module-emulator-st20.c
< +SRC-$(CONFIG_WITH_EMU) += module-emulator-dre2overcrypt.c
< +SRC-$(CONFIG_WITH_EMU) += ffdecsa/ffdecsa.c
< +UNAME := $(shell uname -s)
< +ifneq ($(UNAME),Darwin)
< +ifndef ANDROID_NDK
< +ifndef ANDROID_STANDALONE_TOOLCHAIN
< +ifeq "$(CONFIG_WITH_EMU)" "y"
< +TOUCH_SK := $(shell touch SoftCam.Key)
< +override LDFLAGS += -Wl,--format=binary -Wl,SoftCam.Key -Wl,--format=default
< +endif
< +endif
< +endif
< +endif
<  SRC-$(CONFIG_CS_CACHEEX) += module-csp.c
<  SRC-$(CONFIG_CW_CYCLE_CHECK) += module-cw-cycle-check.c
<  SRC-$(CONFIG_WITH_AZBOX) += module-dvbapi-azbox.c
< @@ -359,7 +383,7 @@
<  # starts the compilation.
<  all:
<  	@./config.sh --use-flags "$(USE_FLAGS)" --objdir "$(OBJDIR)" --make-config.mak
< -	@-mkdir -p $(OBJDIR)/cscrypt $(OBJDIR)/csctapi $(OBJDIR)/minilzo $(OBJDIR)/webif
< +	@-mkdir -p $(OBJDIR)/cscrypt $(OBJDIR)/csctapi $(OBJDIR)/minilzo $(OBJDIR)/ffdecsa $(OBJDIR)/webif
<  	@-printf "\
<  +-------------------------------------------------------------------------------\n\
<  | OSCam ver: $(VER) rev: $(SVN_REV) target: $(TARGET)\n\
149c98
< --- config.h	(revision 11212)
---
> --- config.h	(revision 11086)
161c110
< --- config.sh	(revision 11212)
---
> --- config.sh	(revision 11086)
219c168
< --- cscrypt/md5.c	(revision 11212)
---
[... 671 lines cut ...]
---
> @@ -0,0 +1,18 @@
11438,11445d11295
< +			<TR><TD><A>Process EMM from stream </A></TD>
< +				<TD>
< +					<select name="stream_emm_enabled">
< +						<option value="0" ##STREAMEMMENABLEDSELECTED0##>0 - disabled </option>
< +						<option value="1" ##STREAMEMMENABLEDSELECTED1##>1 - enabled </option>
< +					</select>
< +				</TD>
< +			</TR>
11452a11303
> \ No newline at end of file
11455c11306
< --- webif/files/menu.html	(revision 11212)
---
> --- webif/files/menu.html	(revision 11086)
11482c11333
< --- webif/pages_index.txt	(revision 11212)
---
> --- webif/pages_index.txt	(revision 11086)
11506c11357
< @@ -181,6 +184,7 @@
---
> @@ -180,6 +183,7 @@

closed 2016-06-22 by jahntheman