oscam-emu/oscam-emu #194: Powervu Issue on asiasat 105.5E OpenPli Image

arcan62, 2015-12-12:
TCM,Cartooon Network not opening glitch screen 

The following is the log file.

>> OSCam <<  cardserver log switched at Sat Dec 12 10:45:35 2015
-------------------------------------------------------------------------------
2015/12/12 10:45:35 0542B401 p    (cccam) cccam(r) cs.worldcccam.com: login failed, usr/pwd invalid
2015/12/12 10:45:35 00000000   >> OSCam <<  cardserver log switched, version 1.20-unstable_svn, build r11107 (mipsel-openpli40-linux-gnu-ssl-libusb-pcsc)
2015/12/12 10:45:36 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:39E8D04CDCB0D112DED83BA0C98D8BE3): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:37 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:F320543F6A47A58E9C22ABF07D8CD3B0): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:38 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:EC1E567D01F450D219A9889791F24A9C): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:38 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:981C17EDF55B54BC0D0A2269BD117356): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:39 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:82CB0E4CC7DDE74DE701D266AC4F3B9D): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:40 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:BAE1EA552CA57D9DA8DDD3C5A6A0233D): found (4 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:41 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:78E4BF9C4C3C61C182CB3ED5B26E912F): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:42 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:C16D92C2ECE9C546179A020E3112C146): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:42 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:C41DA682EB7672BD3CCAE8048988EAB1): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:43 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:FBB097DDB8866F4D5B851E8838983F4C): found (4 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:44 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:87420F000003C845DAF493DDA490635A): found (2 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:45 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:07DC5EC9B1986E439960A8D75A274C6B): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:46 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:390C17164F903FF94F58F0A1AE80B385): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:47 0542B401 p   (reader) cs.worldcccam.com [cccam] connecting to worldcccam.com:54000
2015/12/12 10:45:47 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:6B50031A2E9CA1B21DA7AB8A1B2B7D60): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:47 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:5AE804D6725430895CC04F823F1A72F1): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:48 0542B401 p    (cccam) cccam(r) cs.worldcccam.com: login failed, usr/pwd invalid
2015/12/12 10:45:48 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:B47CC256E5102BDE630149609111EB3B): found (3 ms) by oscam_emu (L/1/8/8) - TCM ANZ
2015/12/12 10:45:49 66FF89FA c      (ecm) linuxsatsupport (0E00@000000/0000/0016/40:62A92D326E43677424C18BD248CD9E61): found (4 ms) by oscam_emu (L/1/8/8) - TCM ANZ

arcan62, 2015-12-12:
No one reply yet

budiarno, 2015-12-12:
If you got cw0 & cw1, then the problem is not on the oscam-emu side. It's the image problem.

arcan62, 2015-12-12:
tried on OpenPli and BH but didn't work

arcan62, 2015-12-13:
budiarno what to do know?

arcan62, 2015-12-13:
No Response..... Still Waiting

budiarno, 2015-12-14:
I'm so sorry, I can't help with that, it's beyond my knowledge. May be some one who can do more inside the images can make the picture clear.

arcan62, 2015-12-15:
ok Thanks my dear

jahntheman, 2016-06-22:
some updates have been added.

closed 2016-06-22 by jahntheman

budiarno, 2016-08-24:
Still the same error on PowerVu DES:

> 2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
2016/08/22 22:48:57 00235DDB c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 F0 A0 00 AE A0 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   DD 12 91 F3 63 66 C5 A9 46 49 F3 F1 00 00 20 00 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   00 E8 6B A3 43 87 2B 62 46 E6 FB 6A E6 B5 CA 32 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   02 A9 52 8E C5 EF CE 76 9A 4E 38 0D 66 E1 BB C2 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 F0 A0 00 AE A0 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   DD 12 91 F3 63 66 C5 A9 46 49 F3 F1 00 00 20 00 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   00 E8 6B A3 43 87 2B 62 46 E6 FB 6A E6 B5 CA 32 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   02 A9 52 8E C5 EF CE 76 9A 4E 38 0D 66 E1 BB C2 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 F0 A0 00 AE A0 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   DD 12 91 F3 63 66 C5 A9 46 49 F3 F1 00 00 20 00 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   00 E8 6B A3 43 87 2B 62 46 E6 FB 6A E6 B5 CA 32 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   02 A9 52 8E C5 EF CE 76 9A 4E 38 0D 66 E1 BB C2 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 F2 A0 00 7F 26 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   07 CC 28 8C FB FA F1 BE 57 B5 9F F6 00 00 20 00 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   00 19 35 CE A9 CA 8A F5 FF 72 2B 31 B3 52 91 AE 
2016/08/22 22:48:57 00235DDB c   (dvbapi)   ED D0 52 5B 76 15 49 B2 5C 31 06 55 F2 9E 24 23 
2016/08/22 22:48:57 00235DDB c        (-) -- Skipped 3 duplicated log lines --
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 ECMTYPE 81 CAID 0E00 PROVID 000000 ECMPID 1772 FAKECHID 0000 (unique part in ecm)
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 Filter 2 set ecmtable to EVEN (CAID 0E00 PROVID 000000 FD 13)
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 Filter 2 set chid to ANY CHID on fd 13
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 get controlword!
2016/08/22 22:48:57 00235DDB c      (ecm) Client Busybox ECM dump 0E00@000000/0000/0002/40:B6A457E6DA0972F43349E67A6E05FE57
2016/08/22 22:48:57 00235DDB c      (ecm)   81 30 3D 30 37 20 0E 00 00 00 00 F2 A0 00 7F 26 
2016/08/22 22:48:57 00235DDB c      (ecm)   07 CC 28 8C FB FA F1 BE 57 B5 9F F6 00 00 20 00 
2016/08/22 22:48:57 00235DDB c      (ecm)   00 19 35 CE A9 CA 8A F5 FF 72 2B 31 B3 52 91 AE 
2016/08/22 22:48:57 00235DDB c      (ecm)   ED D0 52 5B 76 15 49 B2 5C 31 06 55 F2 9E 24 23 
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 request controlword for ecm 0E00@000000/0000/0002/40:B6A457E6DA0972F43349E67A6E05FE57
2016/08/22 22:48:57 00235DDB c      (ecm) Client Busybox csphash E4F97CFE cw 000000000000000051C115B607A7A457 rc 0 0E00@000000/0000/0002/40:B6A457E6DA0972F43349E67A6E05FE57
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 controlword received for PID 0 CAID 0E00 PROVID 000000 ECMPID 1772 CHID 0000 VPID 04EC
2016/08/22 22:48:57 00235DDB c   (dvbapi) Demuxer 0 requested controlword for ecm B6A457E6DA0972F43349E67A6E05FE57 on fd 13
2016/08/22 22:48:57 00235DDB c   (dvbapi) generic delay: write cw 60 ms after ecmrequest
2016/08/22 22:48:57 00235DDB c   (dvbapi) delayer: gone=11 ms, cfg=60 ms -> delay=49 ms
2016/08/22 22:48:58 00235DDB c   (dvbapi) Added new streampid 04C4 with index 0 to ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 enable stream 1 pid=0x04c4 index=0 on ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 20)
2016/08/22 22:48:58 00235DDB c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/22 22:48:58 00235DDB c   (dvbapi) Remove streampid 04C4 using indexer 256 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Removed last indexer of streampid 04C4 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/22 22:48:58 00235DDB c   (dvbapi) Added new streampid 04C6 with index 1 to ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 enable stream 2 pid=0x04c6 index=1 on ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 20)
2016/08/22 22:48:58 00235DDB c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/22 22:48:58 00235DDB c   (dvbapi) Remove streampid 04C6 using indexer 257 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Removed last indexer of streampid 04C6 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/22 22:48:58 00235DDB c   (dvbapi) Added new streampid 04C8 with index 2 to ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 enable stream 3 pid=0x04c8 index=2 on ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 20)
2016/08/22 22:48:58 00235DDB c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/22 22:48:58 00235DDB c   (dvbapi) Remove streampid 04C8 using indexer 258 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Removed last indexer of streampid 04C8 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/22 22:48:58 00235DDB c   (dvbapi) Added new streampid 04CA with index 3 to ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 enable stream 4 pid=0x04ca index=3 on ca0
[... cut ...]
2016/08/22 22:48:58 00235DDB c   (dvbapi) Remove streampid 12E1 using indexer 261 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Removed last indexer of streampid 12E1 from ca0
2016/08/22 22:48:58 00235DDB c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/22 22:48:58 00235DDB c      (ecm) Busybox (0E00@000000/0000/0002/40:B6A457E6DA0972F43349E67A6E05FE57): found (8 ms) by emulator - Cartoon Asia

Main error seems on this line:

> CA_SET_PID ioctl error (errno=1 Operation not permitted)