oscam-emu/oscam-emu #206: wrong ecm added for same UA

dodekana, 2016-06-30:
Sony Group on Intelsat 20 and Intelsat 17 sometimes transmit emm data which has the same UA. For example, UA 00422FF6 which is valid for both. Although we already distinguish the Group ID, Oscam doesn't write the new ecm-key to the corresponding Group ID.

oscam-emu, 2017-01-10:
Do you have any logs displaying this issue? Is these channels on C band? I can't receive them.

oscam-emu, 2017-01-22:
Issue confirmed. It is like that by design.... Let me explain.

At the time you tune into a channel, the group id is not known, until a correct/active ECM key is found in any <groupid><sid> combination. **If no active ECM key is found, then groupid is not matched at all.**

At the same time, the EMM algorithm runs, with no matched group id. So if there are 2 or more EMM keys with the same UA in different groups, then each one of them will update the ECM keys in its respective group.

There is no way to fix it. At least easily.

It would be possible to restrict the "wrong" groups updated by checking whether a channel with that sid is present the group. But this not really a fix...

A better solution would be to remove EMM keys with the same UA, at least from groups that have plenty EMM keys.