oscam-emu/oscam-emu #207: PowerVu DES Still Blank on SH4

budiarno, 2016-08-25:
Hello,

I think this still is a problem for use who use sh4 box especially Spark Box. On dvbapi setup with boxtype as dreambox we got this log on OSCam Log:

> 2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
2016/08/25 23:00:18 788500FA c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 EC A0 00 5C F7 
2016/08/25 23:00:18 788500FA c   (dvbapi)   3B 4C C0 8F DB BE 28 C6 F7 B2 55 A0 00 00 10 00 
2016/08/25 23:00:18 788500FA c   (dvbapi)   00 80 55 01 73 77 BE 21 43 4D 56 7C 05 5A 16 6E 
2016/08/25 23:00:18 788500FA c   (dvbapi)   96 18 11 EC 29 BD 1D FF 67 CF 22 5C AA C3 61 E9 
2016/08/25 23:00:18 788500FA c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 EC A0 00 5C F7 
2016/08/25 23:00:18 788500FA c   (dvbapi)   3B 4C C0 8F DB BE 28 C6 F7 B2 55 A0 00 00 10 00 
2016/08/25 23:00:18 788500FA c   (dvbapi)   00 80 55 01 73 77 BE 21 43 4D 56 7C 05 5A 16 6E 
2016/08/25 23:00:18 788500FA c   (dvbapi)   96 18 11 EC 29 BD 1D FF 67 CF 22 5C AA C3 61 E9 
2016/08/25 23:00:18 788500FA c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 EC A0 00 5C F7 
2016/08/25 23:00:18 788500FA c   (dvbapi)   3B 4C C0 8F DB BE 28 C6 F7 B2 55 A0 00 00 10 00 
2016/08/25 23:00:18 788500FA c   (dvbapi)   00 80 55 01 73 77 BE 21 43 4D 56 7C 05 5A 16 6E 
2016/08/25 23:00:18 788500FA c   (dvbapi)   96 18 11 EC 29 BD 1D FF 67 CF 22 5C AA C3 61 E9 
2016/08/25 23:00:18 788500FA c        (-) -- Skipped 2 duplicated log lines --
2016/08/25 23:00:18 788500FA c   (dvbapi) sdt-info dbg: sdt data: 
2016/08/25 23:00:18 788500FA c   (dvbapi)   42 F1 20 00 01 C5 00 00 00 01 FF 00 01 FD 90 12 
2016/08/25 23:00:18 788500FA c   (dvbapi)   48 0C 01 00 09 43 4E 4E 49 20 41 73 69 61 53 02 
2016/08/25 23:00:18 788500FA c   (dvbapi)   0E 00 00 02 FD 90 15 48 0F 01 00 0C 43 61 72 74 
2016/08/25 23:00:18 788500FA c   (dvbapi)   6F 6F 6E 20 41 73 69 61 53 02 0E 00 00 04 FD 90 
2016/08/25 23:00:18 788500FA c   (dvbapi)   17 48 11 01 00 0E 4E 65 77 73 6F 75 72 63 65 20 
2016/08/25 23:00:18 788500FA c   (dvbapi)   44 4E 45 57 53 02 0E 00 00 05 FD 90 17 48 11 01 
2016/08/25 23:00:18 788500FA c   (dvbapi)   00 0E 43 61 72 74 6F 6F 6E 20 54 61 69 77 61 6E 
2016/08/25 23:00:18 788500FA c   (dvbapi)   53 02 0E 00 00 07 FD 90 1C 48 16 01 00 13 43 61 
2016/08/25 23:00:18 788500FA c   (dvbapi)   72 74 6F 6F 6E 20 50 68 69 6C 69 70 70 69 6E 65 
2016/08/25 23:00:18 788500FA c   (dvbapi)   73 53 02 0E 00 00 09 FD 90 10 48 0A 01 00 07 54 
2016/08/25 23:00:18 788500FA c   (dvbapi)   43 4D 20 52 45 47 53 02 0E 00 00 0C FD 90 10 48 
2016/08/25 23:00:18 788500FA c   (dvbapi)   0A 01 00 07 54 4F 4F 4E 41 4D 49 53 02 0E 00 00 
2016/08/25 23:00:18 788500FA c   (dvbapi)   15 FD 90 15 48 0F 01 00 0C 43 4E 20 41 75 73 74 
2016/08/25 23:00:18 788500FA c   (dvbapi)   72 61 6C 69 61 53 02 0E 00 00 16 FD 90 10 48 0A 
2016/08/25 23:00:18 788500FA c   (dvbapi)   01 00 07 54 43 4D 20 41 4E 5A 53 02 0E 00 00 17 
2016/08/25 23:00:18 788500FA c   (dvbapi)   FD 90 16 48 10 01 00 0D 42 6F 6F 6D 65 72 61 6E 
2016/08/25 23:00:18 788500FA c   (dvbapi)   67 20 41 4E 5A 53 02 0E 00 00 C8 FD 90 11 48 0B 
2016/08/25 23:00:18 788500FA c   (dvbapi)   01 00 08 44 6F 77 6E 6C 6F 61 64 53 02 0E 00 9C 
2016/08/25 23:00:18 788500FA c   (dvbapi)   7B E9 F2 
2016/08/25 23:00:18 788500FA c   (dvbapi) sdt-info dbg: iso_mode: -1 offset: 0
2016/08/25 23:00:18 788500FA c   (dvbapi) sdt-info dbg: raw string: 
2016/08/25 23:00:18 788500FA c   (dvbapi)   43 4E 4E 49 20 41 73 69 61 
2016/08/25 23:00:18 788500FA c   (dvbapi) sdt-info dbg: encoded string: 
2016/08/25 23:00:18 788500FA c   (dvbapi)   43 4E 4E 49 20 41 73 69 61 
2016/08/25 23:00:18 788500FA c   (dvbapi) sdt-info (provider:  - channel: CNNI Asia)
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 stop filter 1 try 1 (fd: 12 api: 0, caid: 0001, provid: 000001, emmpid: 0011)
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 flushing stale input data of filter 1 (fd:12)
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
2016/08/25 23:00:18 788500FA c   (dvbapi)   81 30 3D 30 37 20 0E 00 00 00 00 EE A0 00 37 A3 
2016/08/25 23:00:18 788500FA c   (dvbapi)   11 7E F8 E1 8C 7D DE 90 E1 C0 AB E7 00 00 10 00 
2016/08/25 23:00:18 788500FA c   (dvbapi)   00 49 CF 53 A4 B3 10 BE E4 3F 00 54 8C FE 69 1A 
2016/08/25 23:00:18 788500FA c   (dvbapi)   A0 9F 82 E3 53 78 D2 3E A3 E9 34 90 EE FD CF 1E 
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 ECMTYPE 81 CAID 0E00 PROVID 000000 ECMPID 1771 FAKECHID 0000 (unique part in ecm)
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 Filter 2 set ecmtable to EVEN (CAID 0E00 PROVID 000000 FD 13)
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 Filter 2 set chid to ANY CHID on fd 13
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 get controlword!
2016/08/25 23:00:18 788500FA c      (ecm) Client Busybox ECM dump 0E00@000000/0000/0001/40:D242F7434484AA6566342E4C6EBD3421
2016/08/25 23:00:18 788500FA c      (ecm)   81 30 3D 30 37 20 0E 00 00 00 00 EE A0 00 37 A3 
2016/08/25 23:00:18 788500FA c      (ecm)   11 7E F8 E1 8C 7D DE 90 E1 C0 AB E7 00 00 10 00 
2016/08/25 23:00:18 788500FA c      (ecm)   00 49 CF 53 A4 B3 10 BE E4 3F 00 54 8C FE 69 1A 
2016/08/25 23:00:18 788500FA c      (ecm)   A0 9F 82 E3 53 78 D2 3E A3 E9 34 90 EE FD CF 1E 
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 request controlword for ecm 0E00@000000/0000/0001/40:D242F7434484AA6566342E4C6EBD3421
2016/08/25 23:00:18 788500FA c      (ecm) Client Busybox csphash D6D83BE6 cw 00000000000000001AB6E3D3E5AEFB2F rc 0 0E00@000000/0000/0001/40:D242F7434484AA6566342E4C6EBD3421
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 controlword received for PID 0 CAID 0E00 PROVID 000000 ECMPID 1771 CHID 0000 VPID 0488
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 requested controlword for ecm D242F7434484AA6566342E4C6EBD3421 on fd 13
2016/08/25 23:00:18 788500FA c   (dvbapi) generic delay: write cw 60 ms after ecmrequest
2016/08/25 23:00:18 788500FA c   (dvbapi) delayer: gone=12 ms, cfg=60 ms -> delay=48 ms
2016/08/25 23:00:18 788500FA c   (dvbapi) Added new streampid 0460 with index 0 to ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 enable stream 1 pid=0x0460 index=0 on ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 12)
2016/08/25 23:00:18 788500FA c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/25 23:00:18 788500FA c   (dvbapi) Remove streampid 0460 using indexer 256 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Removed last indexer of streampid 0460 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/25 23:00:18 788500FA c   (dvbapi) Added new streampid 0462 with index 1 to ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 enable stream 2 pid=0x0462 index=1 on ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 12)
2016/08/25 23:00:18 788500FA c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/25 23:00:18 788500FA c   (dvbapi) Remove streampid 0462 using indexer 257 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Removed last indexer of streampid 0462 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/25 23:00:18 788500FA c   (dvbapi) Added new streampid 0488 with index 2 to ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 enable stream 3 pid=0x0488 index=2 on ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 12)
2016/08/25 23:00:18 788500FA c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/25 23:00:18 788500FA c   (dvbapi) Remove streampid 0488 using indexer 258 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Removed last indexer of streampid 0488 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/25 23:00:18 788500FA c   (dvbapi) Added new streampid 0939 with index 3 to ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 enable stream 4 pid=0x0939 index=3 on ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Open device /dev/dvb/adapter0/ca0 (fd 12)
2016/08/25 23:00:18 788500FA c   (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)
2016/08/25 23:00:18 788500FA c   (dvbapi) Remove streampid 0939 using indexer 259 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Removed last indexer of streampid 0939 from ca0
2016/08/25 23:00:18 788500FA c   (dvbapi) Demuxer 0 close now unused CA0 device
2016/08/25 23:00:18 788500FA c      (ecm) Busybox (0E00@000000/0000/0001/40:D242F7434484AA6566342E4C6EBD3421): found (10 ms) by emulator - CNNI Asia

OSCam seems unable to set the pid because of the Operation not permitted:
> (dvbapi) CA_SET_PID ioctl error (errno=1 Operation not permitted)

So, in the end none of the pids are decrypted.

harryzz, 2016-09-04:
This is problem in your dvb driver( API for extended CWs not supported ) not oscam-emu. Use StreamRelay as described in readme.

br
harry

budiarno, 2016-09-05:
Well, if that's the problem, then I can do nothing. So far only MCAS is able to decrypt PVU DES correctly on sparkbox (if only it's open source as oscam-emu). Thank you for the explanation.

DeFault-1, 2016-09-18:
I think I have a similar problem using Oscam-Emu on my VU+ clone box with Black Hole image.

Satellite: Asiasat 7 @ 105.5° E

**Sony Pictures Networks India** package on 4180 V is no problem with PowerVu (probably using CSA).

On the other hand **Turner International Asia** package on 3960 H results in black screen (although the correct key is reported to be found in Live Log). Does it mean that **Turner International Asia** package is using DES?

How come the closed source STB boxes never have this kind of problem?

DeFault-1, 2016-09-18:
Stream Relay is working. But the CPU is not fast enough with most receivers to view HD channels.

This is not an option unfortunately.

DeFault-1, 2016-09-18:
This is using DVB API (no Stream Relay).

The PowerVu key is good and it is found by EMU.

No errors in the log.

However there is garbage on the screen.

[http://i.imgur.com/0cFCrGU.jpg](url)

I am using a VU+ clone with Black Hole image.

Maybe the log will tell you something.

 list changed, searching new position
        (-) -- Skipped 1 duplicated log lines --
   (dvbapi) New events occurred on 1 of 3 handlers after 99 ms inactivity
   (dvbapi) Now handling fd 17 that reported event 3
   (dvbapi) Received:
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
 list changed, searching new position
        (-) -- Skipped 1 duplicated log lines --
    (webif) WebIf: Origin checked. Result: access from 192.168.1.38 => allowed
     (work) start reader thread action 13
     (main) starting thread client work
     (main) client work thread started
     (work) data from add_job action=13 client r emulator
   (dvbapi) New events occurred on 1 of 3 handlers after 204 ms inactivity
   (dvbapi) Now handling fd 17 that reported event 3
   (dvbapi) Received:
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
   (dvbapi) New events occurred on 1 of 3 handlers after 198 ms inactivity
   (dvbapi) Now handling fd 17 that reported event 3
   (dvbapi) Received:
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
     (main) starting thread client work
     (main) client work thread started
   (reader) emulator [emu] cardreader_do_checkhealth: reader->card_status = 2, ret = 1
   (dvbapi) New events occurred on 1 of 3 handlers after 203 ms inactivity
   (dvbapi) Now handling fd 17 that reported event 3
   (dvbapi) Received:
   (dvbapi)   80 30 3D 30 37 20 0E 00 00 00 00 37 A0 00 80 CA 
   (dvbapi)   AE F0 B0 50 E0 CB EB F2 60 A1 69 B5 00 00 20 00 
   (dvbapi)   00 99 CB C0 70 94 AD 26 34 32 D3 C0 3C B6 FF EF 
   (dvbapi)   DB 85 D5 2B 22 A7 9C 9E 03 81 BA 16 A5 A2 85 94 
   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x040):
[... 75 lines cut ...]
   (dvbapi) Demuxer 0 write cw0 index: 0 (ca0)
   (dvbapi) Demuxer 0 ca0 is using index 1 for streampid 04C8 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 2 for streampid 04CA -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 3 for streampid 04EC -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 0 for streampid 04C4 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 1 for streampid 04C8 -> write odd part of cw!
   (dvbapi) Demuxer 0 writing odd part (8CFDFD377A45F17F) of controlword, replacing expired (8CFDFD377A45F17F)
   (dvbapi) Demuxer 0 write cw0 index: 1 (ca0)
   (dvbapi) Demuxer 0 ca0 is using index 2 for streampid 04CA -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 3 for streampid 04EC -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 0 for streampid 04C4 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 1 for streampid 04C8 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 2 for streampid 04CA -> write odd part of cw!
   (dvbapi) Demuxer 0 writing odd part (8CFDFD377A45F17F) of controlword, replacing expired (8CFDFD377A45F17F)
   (dvbapi) Demuxer 0 write cw0 index: 2 (ca0)
   (dvbapi) Demuxer 0 ca0 is using index 3 for streampid 04EC -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 0 for streampid 04C4 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 1 for streampid 04C8 -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 2 for streampid 04CA -> skip!
   (dvbapi) Demuxer 0 ca0 is using index 3 for streampid 04EC -> write odd part of cw!
   (dvbapi) Demuxer 0 writing odd part (F1C2A19D19A2D37F) of controlword, replacing expired (8CFDFD377A45F17F)
   (dvbapi) Demuxer 0 write cw0 index: 3 (ca0)
      (ecm) DeFault (0E00@000000/0000/0002/40:ED49DCAE2FDFEC2A2800B05F23CA9C02): found (54 ms) by emulator - Cartoon Asia
      (ecm) cw:
      (ecm)   F1 C2 A1 9D 19 A2 D3 7F 00 00 00 00 00 00 00 00

budiarno, 2017-04-02:
Not oscam issue related. Thanks to Joeuser for the fixes in Enigma2 drivers.

closed 2017-04-02 by budiarno

DeFault-1, 2017-04-03:
Wow, good news!

Where is the patched driver?

budiarno, 2017-04-04:
Joeuser posted the image patched for the hardware DES on Linuxsat-Support forum.

DeFault-1, 2017-04-04:
Very good. Thank you.