oscam-emu/oscam-emu #211: can't compile with the last oscam version r11278

messi89, 2016-10-22:
Hi

can you update to r11278 ?

messi89, 2016-10-22:
`CC	module-emulator-osemu.c
module-emulator-osemu.c: In function ‘PowervuECM’:
module-emulator-osemu.c:3033:11: error: ‘EXTENDED_CW’ has no member named ‘mode’
      cw_ex->mode = CW_MODE_MULTIPLE_CW;
           ^
module-emulator-osemu.c:3037:12: error: ‘EXTENDED_CW’ has no member named ‘algo’
       cw_ex->algo = CW_ALGO_CSA;
            ^
module-emulator-osemu.c:3038:12: error: ‘EXTENDED_CW’ has no member named ‘algo_mode’
       cw_ex->algo_mode = CW_ALGO_MODE_ECB;
            ^
module-emulator-osemu.c:3042:12: error: ‘EXTENDED_CW’ has no member named ‘algo’
       cw_ex->algo = CW_ALGO_DES;
            ^
module-emulator-osemu.c:3043:12: error: ‘EXTENDED_CW’ has no member named ‘algo_mode’
       cw_ex->algo_mode = CW_ALGO_MODE_ECB;
            ^
module-emulator-osemu.c:3048:18: error: ‘EXTENDED_CW’ has no member named ‘audio’
       dwp = cw_ex->audio[j];
                  ^
module-emulator-osemu.c:3078:17: error: ‘EXTENDED_CW’ has no member named ‘data’
      dwp = cw_ex->data;
                 ^`

FCS001FCS, 2016-10-22:
For me EMU Patch works up to OSCAM SVN 11276. It fails to compile from 11277 and up to current SVN 11279.
I use OSCAM SimpleBuild to compile on Ubuntu 16.04 on a VM in a Win 10 host.

M3mphis, 2016-10-22:
from 
#if defined(READER_DRE) || defined(READER_DRECAS)
to
#if defined(READER_DRE) || defined(READER_DRECAS) || defined(READER_VIACCESS)

ciro73, 2016-10-23:
and you need to remove the space on line 2932 after the last defined statement
```
2932c2932
<  #if defined(READER_DRE) || defined(READER_DRECAS) || defined (READER_VIACCESS)
---
>  #if defined(READER_DRE) || defined(READER_DRECAS) || defined(READER_VIACCESS)
```

fabioganga, 2016-11-01:
I'm still unable to compile with 11279 for MIPSOE20, despite having followed the above suggestions from M3mphis.

My build.log shows the following (I'm just pasting the last few lines):

> module-emulator-osemu.c:8659:15: note: previous definition of 'Irdeto2EMM' was here
module-emulator-osemu.c:13467:9: error: redefinition of 'GetIrdeto2Hexserial'
module-emulator-osemu.c:4053:9: note: previous definition of 'GetIrdeto2Hexserial' was here
module-emulator-osemu.c:13502:15: error: redefinition of 'PowervuEMM'
module-emulator-osemu.c:8795:15: note: previous definition of 'PowervuEMM' was here
module-emulator-osemu.c:13595:9: error: redefinition of 'GetPowervuHexserials'
module-emulator-osemu.c:4181:9: note: previous definition of 'GetPowervuHexserials' was here
module-emulator-osemu.c:13660:15: error: redefinition of 'GetDrecryptEMMKey'
module-emulator-osemu.c:8953:15: note: previous definition of 'GetDrecryptEMMKey' was here
module-emulator-osemu.c:13667:13: error: redefinition of 'DrecryptWriteEEToFile'
module-emulator-osemu.c:8960:13: note: previous definition of 'DrecryptWriteEEToFile' was here
module-emulator-osemu.c:13680:15: error: redefinition of 'DrecryptProcessEMM'
module-emulator-osemu.c:8973:15: note: previous definition of 'DrecryptProcessEMM' was here
module-emulator-osemu.c:13804:15: error: redefinition of 'Drecrypt2EMM'
module-emulator-osemu.c:9097:15: note: previous definition of 'Drecrypt2EMM' was here
module-emulator-osemu.c:13845:9: error: redefinition of 'GetDrecryptHexserials'
module-emulator-osemu.c:4431:9: note: previous definition of 'GetDrecryptHexserials' was here
module-emulator-osemu.c:13880:15: error: redefinition of 'GetTandbergEMMKey'
module-emulator-osemu.c:9173:15: note: previous definition of 'GetTandbergEMMKey' was here
module-emulator-osemu.c:13885:15: error: redefinition of 'TandbergParseEMMNanoTags'
module-emulator-osemu.c:9178:15: note: previous definition of 'TandbergParseEMMNanoTags' was here
module-emulator-osemu.c:13982:15: error: redefinition of 'TandbergParseEMMNanoData'
module-emulator-osemu.c:9275:15: note: previous definition of 'TandbergParseEMMNanoData' was here
module-emulator-osemu.c:14010:15: error: redefinition of 'TandbergEMM'
module-emulator-osemu.c:9303:15: note: previous definition of 'TandbergEMM' was here
module-emulator-osemu.c:14060:13: error: redefinition of 'GetProcessEMMErrorReason'
module-emulator-osemu.c:4646:13: note: previous definition of 'GetProcessEMMErrorReason' was here
module-emulator-osemu.c:14088:8: error: redefinition of 'ProcessEMM'
module-emulator-osemu.c:4674:8: note: previous definition of 'ProcessEMM' was here
make[1]: *** [build/mipsel-unknown-linux-gnu/module-emulator-osemu.o] Error 1
make: *** [all] Error 2

> remove patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
Hunk #1 succeeded at 1 with fuzz 1.
patching file config.sh
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_064_long.h
patching file ffdecsa/parallel_128_sse2.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
patching file Makefile
patching file module-dvbapi.c
patching file module-dvbapi.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-newcamd.c
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
patching file oscam-chk.c
patching file oscam-config-global.c
patching file oscam-config-reader.c
patching file oscam-config.h
patching file oscam-ecm.c
patching file oscam-emm.c
patching file oscam-string.c
patching file oscam-string.h
patching file oscam.c
patching file reader-common.c
patching file readers.h
patching file webif/config/dvbapi.html
patching file webif/config/menu.html
patching file webif/config/menu_streamrelay.html
patching file webif/config/streamrelay.html
Unreversed patch detected!  Ignore -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file webif/config/streamrelay.html.rej
patching file webif/files/menu.html
patching file webif/files/menu_softcamkey.html
patching file webif/pages_index.txt
patching file webif/readerconfig/readerconfig_emubit.html

> delete  oscam-emu.patch file from source

Is there anything else I can try to fix this, please?

Thanks!

commando2015, 2016-11-23:
Thanks @M3mphis !

nautilus7, 2017-01-05:
fixed compilation against svn 11313 in pullrequest #215

closed 2017-01-10 by oscam-emu