oscam-emu/oscam-emu #217: Recording & Watching - Cannot Decode 2 PowerVu Channels Simutanously - OSCAM EMU

FCS001FCS, 2016-12-18:
This was posted at "linuxsat-support.com" but I thought it might get an answer here from developers of the EMU.
+++++++++++++++
Hardware: Dreambox DM7020HD (mips32el) - OpenPLi 4.0 and OSCAM-EMU r11281

With the dual tuner DM7020HD, when I am recording a PowerVu channel on Tuner B and I try and view another PowerVu channel on the same transponder (different SID), I get a black screen.

It happens the same in reverse, if I am watching a PowerVu channel and a timer recording starts on the same transponder, I get a black screen recording.

This is all direct decode, no stream relay involved on the DM7020HD.

I am informed by an EMU developer that it is normal that the DM7020HD does not switch to its other Tuner (A in the above example) since it is remaining on the same Sat and Transponder and should not need to switch.

The EMU developer also mentioned that it is most likely due to the number of PIDs that are allowed to be processed by OSCAM-EMU and that I should post the problem in public so that everyone can help and get the benefit, so I am.

When I switch to one of the PowerVu channels I get the following info concerning PIDS:

Code:
```
2016/12/17 00:06:17 xxxxxxxx c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 0E00 ECM_PID: 1772 PROVID: 000000 
2016/12/17 00:06:17 xxxxxxxx c   (dvbapi) Demuxer 0 found 1 ECMpids and 10 STREAMpids in caPMT
2016/12/17 00:06:17 xxxxxxxx c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1772 ANY CHID PMTPID 138A VPID 04EC
2016/12/17 00:06:17 xxxxxxxx c   (dvbapi) Demuxer 0 found 1 ECMpids and 10 STREAMpids in PMT
```

The STREAMpids are usually 10 but sometimes 9 and 1 time is 5 for a HD channel.

I have no idea what is needed so let me know if a log is needed and at what debug level it should be taken in order to understand the problem and hopefully develop a fix.

egydoctor2010, 2017-11-22:
you can try it after pid limiter options have been added to the emu make it only 5 stream pids at first and try recording if unsucessful try to limit more pids

FCS001FCS, 2017-11-23:
Thanks. Will give it a try.

oscam-emu, 2017-12-02:
@FCS001FCS did you try the pid limiter? Does it fixed the problem?

Btw, as of r760 the pid limiter is only applied to powervu channels, and not to all caids.

FCS001FCS, 2017-12-03:
Not yet.

I have not compiled OSCAM and EMU in a long time so will have to learn how to do it again.

FCS001FCS, 2017-12-03:
@oscam-emu

I had a hard time trying to figure out how to cross-compile on a Ubuntu 17 VM on Windows 10 PC for MIPSEL OpenPli 4 OSCAM-EMU so I just downloaded a binary from [http://www.streamboard.tv/wbb2/thread.php?threadid=27014&page=166](url) from user @ sattotal. File is: [http://www.streamboard.tv/wbb2/attachment.php?attachmentid=67460](url) (oscam-svn11392-openpli40-webif-dvbapi-ssl-libusb-emu-upx). It has the R760 EMU included.

I set "Max pids for extended CWs" in the "DVB-API" config tab in GUI to "5" (was set at "64"). Then OSCAM restart.

Success. I can now watch an AFN channel with direct decode (not stream relay) and also record on another AFN channel on my DM7020HD with OpenPli 4.0.

Many Thanks to  @oscam-emu and @egydoctor2010 for your help and hard work.

[image]

oscam-emu, 2017-12-03:
Ok, nice.

The problem you have is caused by the drivers of your box and not the emu iteself, but anyway. That's why joeuser made this pid limiter patch and we decided to include it in emu as well.

You can easily cross compile oscam-emu with simplebuild 3. You can find it in the streamboard forum. It works on ubuntu on windows 10 just fine. No need for a VM. https://msdn.microsoft.com/en-us/commandline/wsl/about

closed 2017-12-03 by oscam-emu