oscam-emu/oscam-emu #222: rev 11310

mavroskoufis, 2017-01-03:
I tray to compile with 
make CROSS=$TOOLCHAIN/bin/mipsel-tuxbox-linux-gnu- USE_SSL=1
and have this ERROR
module-webif.c: In function 'send_oscam_files':
module-webif.c:6296:24: error: 'MNU_CFG_FSOFTCAMKEY' undeclared (first use in this function)
   { "SoftCam.Key",     MNU_CFG_FSOFTCAMKEY,FTYPE_CONFIG }, // id 19

module-webif.c:6296:24: note: each undeclared identifier is reported only once for each function it appears in
Makefile:437: recipe for target 'build/mipsel-tuxbox-linux-gnu-ssl/module-webif.o' failed
make[1]: *** [build/mipsel-tuxbox-linux-gnu-ssl/module-webif.o] Error 1
Makefile:389: recipe for target 'all' failed
make: *** [all] Error 2
THANKS

sculpneo, 2017-01-03:
same issue

sculpneo, 2017-01-06:
it works, thank you

mavroskoufis, 2017-01-06:
No for me with 
oscam-emu.patch
same error

nautilus7, 2017-01-06:
You're are not using my patch from the pull request.

This is the file:  https://github.com/nautilus7/oscam-emu/blob/tandberg-ecm-emm-fix/oscam-emu.patch

sculpneo, 2017-01-06:
https://github.com/nautilus7/oscam-emu/archive/tandberg-ecm-emm-fix.zip

mavroskoufis, 2017-01-08:
Thanks Works !!!!

closed 2017-01-10 by oscam-emu