oscam-emu/oscam-emu #226: error compilation

messi89, 2017-01-13:
`
appliquer patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_064_long.h
patching file ffdecsa/parallel_128_sse2.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
patching file Makefile
patching file module-dvbapi.c
patching file module-dvbapi.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-newcamd.c
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
patching file oscam-chk.c
patching file oscam-config-global.c
patching file oscam-config-reader.c
patching file oscam-config.h
patching file oscam-ecm.c
patching file oscam-emm.c
patching file oscam-string.c
patching file oscam-string.h
patching file oscam.c
patching file reader-common.c
patching file readers.h
patching file webif/config/dvbapi.html
patching file webif/config/menu.html
patching file webif/config/menu_streamrelay.html
patching file webif/config/streamrelay.html
patching file webif/files/menu.html
patching file webif/files/menu_softcamkey.html
patching file webif/pages_index.txt
patching file webif/readerconfig/readerconfig_emubit.html

+-------------------------------------------------------------------------------
| OSCam ver: 1.20-unstable_svn rev: 11330 target: x86_64-linux-gnu
| Tools:
|  CROSS    = 
|  CC       = gcc
| Settings:
|  CONF_DIR = /usr/local/etc
|  CC_OPTS  = -Os -ggdb -pipe -ffunction-sections -fdata-sections
|  CC_WARN  = -w
|  CFLAGS   = -fexpensive-optimizations -mmmx -msse -msse2 -msse3
|  LDFLAGS  = -Wl,--gc-sections -lz -Wl,-rpath,/usr/lib
|  LIBS     = -lpthread -ldl -lrt
|  UseFlags = 
| Config:
|  Addons   : WEBIF WEBIF_LIVELOG WEBIF_JQUERY TOUCH READ_SDT_CHARSETS IRDETO_GUESSING CS_ANTICASC WITH_DEBUG MODULE_MONITOR WITH_LB CS_CACHEEX CW_CYCLE_CHECK CLOCKFIX WITH_EMU
[... 233 lines cut ...]
                ^
module-emulator.c:754:16: note: previous definition of ‘EMU_card_write’ was here
 static int32_t EMU_card_write(struct s_reader *UNUSED(pcsc_reader),const uchar *UNUSED(buf) ,uint8_t *UNUSED(cta_res),
                ^
module-emulator.c:1600:16: error: redefinition of ‘EMU_set_protocol’
 static int32_t EMU_set_protocol(struct s_reader *UNUSED(rdr),uint8_t *UNUSED(params),uint32_t *UNUSED(length),
                ^
module-emulator.c:756:16: note: previous definition of ‘EMU_set_protocol’ was here
 static int32_t EMU_set_protocol(struct s_reader *UNUSED(rdr),uint8_t *UNUSED(params),uint32_t *UNUSED(length),
                ^
module-emulator.c:1603:27: error: redefinition of ‘cardreader_emu’
 const struct s_cardreader cardreader_emu =
                           ^
module-emulator.c:759:27: note: previous definition of ‘cardreader_emu’ was here
 const struct s_cardreader cardreader_emu =
                           ^
module-emulator.c:1619:6: error: redefinition of ‘add_emu_reader’
 void add_emu_reader(void)
      ^
module-emulator.c:775:6: note: previous definition of ‘add_emu_reader’ was here
 void add_emu_reader(void)
      ^
make[1]: *** [build/x86_64-linux-gnu/module-emulator.o] Erreur 1
make[1]: *** Attente des tâches non terminées....
make: *** [all] Erreur 2`

messi89, 2017-01-13:
oscam 11330

nautilus7, 2017-01-13:
Something is wrong on your part. Works fine here.

messi89, 2017-01-13:
@nautilus7 I will try to clean and checkout oscam again.

messi89, 2017-01-13:
fixed by clean/checkout oscam

nautilus7, 2017-01-13:
ok thanks.

closed 2017-01-13 by nautilus7