oscam-emu/oscam-emu #227: new patch

beppeniosi, 2017-01-14:
sorry, I need help!!!
I cant compile with simplebuild, not with 11330 or 11334. everytime I've errors. I attach log.
sorry for my bad english...
thanks in advance

[build.zip](https://github.com/oscam-emu/oscam-emu/files/706067/build.zip)

ceco2000, 2017-01-14:
Oscam version of the trunk you download?

beppeniosi, 2017-01-14:
I use ./simplebuild menu in my linux server. I usually put the patch in the right dir: /root/simple/patches and I compile like I've ever done.
I deleted all config and re-checked out the new ones...but nothing to do!!!
Tnx for your help

ceco2000, 2017-01-14:
Well, in the folder simplebuild / if you use version 2 simplebuild / or / s3 if you use the new simplebuild / delete folder oscam-svn. Start simplebuild left to check and download the latest svn trunk. Having completed the exit from simplebuild, open the folder oscam-svn located in the folder simplebuild or s3, copy oscam-emu.patch file and put it in a folder oscam-svn. Start terminal and write

> cd $HOME/simplebuild/oscam-svn 
> or
> cd $HOME/s3/oscam-svn
> print enter
> then write
> patch -p0 < oscam-emu.patch
> print enter

Then run simplebuild, and can build up. But for svn.11334 will have to wait for repairing oscam-emu.patch.

udz99, 2017-01-14:
same problem
im using s3 compile build tools, error while compile oscam-svn11334, OK, waiting for repairing oscam-emu.patch......

beppeniosi, 2017-01-14:
@ceco2000 
I'tryed, but it doesnt work: device reader error
Wait for the new fix
tnx

ceco2000, 2017-01-14:
For now you get a chance to experiment with svn.11330. svn.11335 patching will be made after the necessary changes, and declared in Github oscam-emu.

closed 2017-01-14 by nautilus7