oscam-emu/oscam-emu #228: error patch oscam-svn11334

udz99, 2017-01-14:
build with no emu working fine, after patched with emu error build
**Lastbuild.log**

    ___  ____   ___
   / _ \/ ___| / __|__ _ _ __ ___
  | | | \___ \| |  / _` | '_ ` _ \ 
  | |_| |___) | |_| (_| | | | | | |
   \___/|____/ \___\__,_|_| |_| |_|
   
+-------------------------------------------------------------------------------
| OSCam ver: 1.20-unstable_svn rev: 11334 target: mipsel-openpli40-linux-gnu-libusb
| Tools:
|  CROSS    = /opt/s6/support/toolchains/openpli40/bin/mipsel-openpli40-linux-gnu-
|  CC       = /opt/s6/support/toolchains/openpli40/bin/mipsel-openpli40-linux-gnu-gcc
| Settings:
|  CONF_DIR = /usr/keys
|  CC_OPTS  = -O0 -ggdb -pipe -ffunction-sections -fdata-sections -finline-functions -fno-short-enums -freg-struct-return
|  CC_WARN  = -W -Wall -Wshadow -Wredundant-decls -Wstrict-prototypes -Wold-style-definition
|  CFLAGS   = -fexpensive-optimizations -DWITH_LIBCRYPTO=1 -DWITH_LIBUSB=1 -DWITH_UTF8=1 -DWITH_UTF8
|  LDFLAGS  = -Wl,--gc-sections -DWITH_LIBCRYPTO=1 -DWITH_LIBUSB=1 -DWITH_UTF8=1 -DWITH_UTF8
|  LIBS     = -lcrypto -lusb-1.0 -lrt -lpthread -ldl -lrt
|  UseFlags = USE_LIBCRYPTO=1 USE_LIBUSB=1 USE_UTF8=1
| Config:
|  Addons   : WEBIF WEBIF_LIVELOG WEBIF_JQUERY TOUCH HAVE_DVBAPI READ_SDT_CHARSETS IRDETO_GUESSING CS_ANTICASC WITH_DEBUG MODULE_MONITOR WITH_LB CS_CACHEEX CW_CYCLE_CHECK CLOCKFIX WITH_EMU
|  Protocols: CAMD35 CAMD35_TCP NEWCAMD CCCAM CCCSHARE GBOX RADEGAST SCAM SERIAL CONSTCW PANDORA GHTTP
|  Readers  : NAGRA IRDETO CONAX CRYPTOWORKS SECA VIACCESS VIDEOGUARD DRE TONGFANG BULCRYPT GRIFFIN DGCRYPT
|  CardRdrs : PHOENIX INTERNAL SC8IN1 MP35 SMARGO DB2COM STAPI STINGER DRECAS SMART
|  Compiler : mipsel-openpli40-linux-gnu-gcc (crosstool-NG hg+default-86a8d1d467c8) 4.8.1
|  Config   : build/mipsel-openpli40-linux-gnu-libusb/config.mak
|  Binary   : /opt/s6/support/binaries/oscam-svn11334-openpli40-webif-dvbapi-libusb-emu
+-------------------------------------------------------------------------------
HOSTCC	webif/pages_gen
GEN	webif/pages.c
GEN	Compressed 435512 template bytes into 177066 bytes. 258446 saved bytes (59.34%).
CC	cscrypt/i_cbc.c
CC	cscrypt/i_ecb.c
CC	cscrypt/des.c
CC	cscrypt/i_skey.c
CC	cscrypt/rc6.c
CC	csctapi/ifd_sc8in1.c
CC	csctapi/protocol_t0.c
CC	csctapi/ifd_db2com.c
CC	csctapi/ifd_drecas.c
CC	csctapi/ifd_mp35.c
CC	csctapi/ifd_phoenix.c
CC	cscrypt/md5.c
CC	csctapi/icc_async.c
CC	csctapi/ifd_sci.c
CC	csctapi/atr.c
CC	csctapi/protocol_t1.c
CC	csctapi/io_serial.c
CC	csctapi/ifd_smargo.c
CC	csctapi/ifd_smartreader.c
CC	csctapi/ifd_stinger.c
CC	minilzo/minilzo.c
CC	module-anticasc.c
CC	module-cacheex.c
CC	module-camd35-cacheex.c
CC	module-camd35.c
CC	module-cccam-cacheex.c
CC	module-cccam.c
CC	module-constcw.c
CC	module-cccshare.c
CC	module-emulator.c
CC	ffdecsa/ffdecsa.c
CC	module-emulator-stream.c
CC	module-emulator-osemu.c
CC	module-csp.c
CC	module-cw-cycle-check.c
CC	module-dvbapi-chancache.c
CC	module-gbox-helper.c
CC	module-dvbapi.c
CC	module-gbox-sms.c
CC	module-gbox-cards.c
CC	module-gbox.c
CC	module-ird-guess.c
CC	module-monitor.c
CC	module-newcamd-des.c
CC	module-ghttp.c
CC	module-newcamd.c
CC	module-pandora.c
CC	module-radegast.c
CC	module-scam.c
CC	module-serial.c
CC	module-stat.c
CC	module-webif-lib.c
CC	module-webif-tpl.c
CC	module-webif.c
CC	reader-common.c
CC	webif/pages.c
CC	reader-bulcrypt.c
CC	reader-conax.c
CC	reader-cryptoworks.c
CC	reader-dgcrypt.c
CC	reader-dre.c
CC	reader-dre-cas.c
CC	reader-dre-common.c
CC	reader-dre-st20.c
CC	reader-griffin.c
CC	reader-irdeto.c
CC	reader-nagra.c
CC	reader-seca.c
CC	reader-tongfang.c
CC	reader-viaccess.c
CC	reader-videoguard-common.c
CC	reader-videoguard1.c
CC	reader-videoguard12.c
CC	oscam-hashtable.c
CC	oscam-aes.c
CC	reader-videoguard2.c
CC	oscam-array.c
CC	oscam-chk.c
CC	oscam-cache.c
CC	oscam-client.c
CC	oscam-conf.c
CC	oscam-conf-mk.c
CC	oscam-conf-chk.c
CC	oscam-config-account.c
CC	oscam-config-global.c
CC	oscam-config.c
CC	oscam-config-reader.c
CC	oscam-ecm.c
CC	oscam-emm.c
CC	oscam-emm-cache.c
oscam-config-reader.c: In function 'ftab_fn':
oscam-config-reader.c:453:18: error: 'FTAB_EMUAU' undeclared (first use in this function)
   if(ftab_type & FTAB_EMUAU)       { rdr = container_of(setting, struct s_reader, emu_auproviders); }
                  ^
oscam-config-reader.c:453:18: note: each undeclared identifier is reported only once for each function it appears in
In file included from oscam-config-reader.c:8:0:
oscam-config-reader.c: At top level:
oscam-config-reader.c:907:93: error: 'FTAB_EMUAU' undeclared here (not in a function)
  DEF_OPT_FUNC_X("emu_auproviders"    , OFS(emu_auproviders),         ftab_fn, FTAB_READER | FTAB_EMUAU),
                                                                                             ^
oscam-conf.h:116:18: note: in definition of macro 'DEF_OPT_FUNC_X'
   .def.d_extra = __extra, \
                  ^
Makefile:437: recipe for target 'build/mipsel-openpli40-linux-gnu-libusb/oscam-config-reader.o' failed
make[1]: *** [build/mipsel-openpli40-linux-gnu-libusb/oscam-config-reader.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC	oscam-failban.c
CC	oscam-files.c
CC	oscam-garbage.c
Makefile:389: recipe for target 'all' failed
make: *** [all] Error 2
TIME	[  0 min(s) 6 secs ]

 Filesize before     	= 0.00k
 Filesize after      	= 0.00k
 Filesize reduced to 	=

ceco2000, 2017-01-14:
Leave nautilus7 peace will quickly raise his hands, and might give up the support of emu.

udz99, 2017-01-14:
ok.....will wait....thanks

closed 2017-01-14 by nautilus7