oscam-emu/oscam-emu #230: Can't patch latest revision 11347

genox87, 2017-01-23:
Hello, i've always get issue while i try to patch and compile against the latest revision of oscam ( 11347 )

> Ripristinato 'module-webif-tpl.h'
> Ripristinato 'oscam-array.h'
> Ripristinato 'oscam-conf.c'
> Alla revisione 11347.

Syncd the svn with the latest relase of oscam 

> user@vps:~/oscam-svn$ svn patch oscam-emu.patch
> U         CMakeLists.txt
> U         config.h
> U         config.sh
> U         cscrypt/md5.c
> U         cscrypt/md5.h
> U         csctapi/cardreaders.h
> Saltato oggetto mancante: 'ffdecsa/CMakeLists.txt'
> Saltato oggetto mancante: 'ffdecsa/COPYING'
> Saltato oggetto mancante: 'ffdecsa/Makefile'
> Saltato oggetto mancante: 'ffdecsa/ffdecsa.c'
> Saltato oggetto mancante: 'ffdecsa/ffdecsa.h'
> Saltato oggetto mancante: 'ffdecsa/fftable.h'
> Saltato oggetto mancante: 'ffdecsa/parallel_032_int.h'
> Saltato oggetto mancante: 'ffdecsa/parallel_064_long.h'
> Saltato oggetto mancante: 'ffdecsa/parallel_128_sse2.h'
> Saltato oggetto mancante: 'ffdecsa/parallel_generic.h'
> Saltato oggetto mancante: 'ffdecsa/parallel_std_def.h'
> Saltato oggetto mancante: 'ffdecsa/stream.c'
> U         globals.h
> U         Makefile
> U         module-dvbapi.c
> U         module-dvbapi.h
> Saltato oggetto mancante: 'module-emulator-osemu.c'
> Saltato oggetto mancante: 'module-emulator-osemu.h'
> Saltato oggetto mancante: 'module-emulator-stream.c'
> Saltato oggetto mancante: 'module-emulator-stream.h'
> Saltato oggetto mancante: 'module-emulator.c'
> U         module-newcamd-des.c
> U         module-newcamd-des.h
> U         module-newcamd.c
> U         module-stat.c
> U         module-webif-tpl.c
> U         module-webif.c
> U         oscam-chk.c
> U         oscam-config-global.c
> U         oscam-config-reader.c
> U         oscam-config.h
> U         oscam-ecm.c
> U         oscam-emm.c
> U         oscam-string.c
> U         oscam-string.h
> U         oscam.c
> U         reader-common.c
> U         readers.h
> U         webif/config/dvbapi.html
> U         webif/config/menu.html
> Saltato oggetto mancante: 'webif/config/menu_streamrelay.html'
> Saltato oggetto mancante: 'webif/config/streamrelay.html'
> U         webif/files/menu.html
> Saltato oggetto mancante: 'webif/files/menu_softcamkey.html'
> U         webif/pages_index.txt
> Saltato oggetto mancante: 'webif/readerconfig/readerconfig_emubit.html'
> Summary of conflicts:
> Skipped paths: 21
> 

I've removed all the files, and started with a working copy of oscam, but when i try to patch i always gettings errors.

oscam-emu, 2017-01-23:
You're doing something wrong, but I don't speak Italian so I can't help you.

Try to delete oscam-svn folder and download the patch file again.

genox87, 2017-01-24:
* Edit: rm'ing the whole folder did the trick, i've rm-rf the whole oscam-svn and then svn checkout http://www.streamboard.tv/svn/oscam/trunk oscam-svn

Thanks

Well, the translation it's pretty simple, 'Saltato oggetto mancante: ' = skip missing object...

Of course before fill the report , i've "rm -rf *" the content of the folder "oscam-svn" and then "svn update" to get the sync to the latest revision from oscam svn.

closed 2017-01-24 by genox87