oscam-emu/oscam-emu #232: compile error 11355

sculpneo, 2017-02-09:
apply patch oscam-emu.patch
patching file CMakeLists.txt
patching file config.h
patching file config.sh
patching file cscrypt/md5.c
patching file cscrypt/md5.h
patching file csctapi/cardreaders.h
patching file ffdecsa/CMakeLists.txt
patching file ffdecsa/COPYING
patching file ffdecsa/Makefile
patching file ffdecsa/ffdecsa.c
patching file ffdecsa/ffdecsa.h
patching file ffdecsa/fftable.h
patching file ffdecsa/parallel_032_int.h
patching file ffdecsa/parallel_064_long.h
patching file ffdecsa/parallel_128_sse2.h
patching file ffdecsa/parallel_generic.h
patching file ffdecsa/parallel_std_def.h
patching file ffdecsa/stream.c
patching file globals.h
patching file Makefile
patching file module-dvbapi.c
Hunk #6 succeeded at 6231 (offset 3 lines).
patching file module-dvbapi.h
patching file module-emulator-osemu.c
patching file module-emulator-osemu.h
patching file module-emulator-stream.c
patching file module-emulator-stream.h
patching file module-emulator.c
patching file module-newcamd-des.c
patching file module-newcamd-des.h
patching file module-newcamd.c
patching file module-stat.c
patching file module-webif-tpl.c
patching file module-webif.c
Hunk #2 FAILED at 132.
Hunk #3 succeeded at 1212 (offset 2 lines).
Hunk #4 succeeded at 1655 (offset 2 lines).
Hunk #5 succeeded at 2120 (offset 2 lines).
Hunk #6 succeeded at 2698 (offset 2 lines).
Hunk #7 succeeded at 2734 (offset 2 lines).
Hunk #8 succeeded at 4464 (offset 2 lines).
Hunk #9 succeeded at 4971 (offset 2 lines).
Hunk #10 succeeded at 6417 with fuzz 2 (offset 4 lines).
Hunk #11 succeeded at 6890 (offset 4 lines).
Hunk #12 succeeded at 7897 (offset 4 lines).
1 out of 12 hunks FAILED -- saving rejects to file module-webif.c.rej
patching file oscam-chk.c
patching file oscam-config-global.c
patching file oscam-config-reader.c
patching file oscam-config.h
patching file oscam-ecm.c
patching file oscam-emm.c
patching file oscam-string.c
patching file oscam-string.h
patching file oscam.c
patching file reader-common.c
patching file readers.h
patching file webif/config/dvbapi.html
patching file webif/config/menu.html
patching file webif/config/menu_streamrelay.html
patching file webif/config/streamrelay.html
patching file webif/files/menu.html
Hunk #1 FAILED at 10.
1 out of 2 hunks FAILED -- saving rejects to file webif/files/menu.html.rej
patching file webif/files/menu_softcamkey.html
patching file webif/pages_index.txt
patching file webif/readerconfig/readerconfig_emubit.html

+-------------------------------------------------------------------------------
| OSCam ver: 1.20-unstable_svn rev: 11355 target: mipsel-openpli40-linux-gnu
| Tools:
|  CROSS    = /home/mert/mert/toolchains/openpli40/bin/mipsel-openpli40-linux-gnu-
|  CC       = /home/mert/mert/toolchains/openpli40/bin/mipsel-openpli40-linux-gnu-gcc
| Settings:
|  CONF_DIR = /etc/tuxbox/config
|  CC_OPTS  = -Os -ggdb -pipe -ffunction-sections -fdata-sections
|  CC_WARN  = -w
|  CFLAGS   = -fexpensive-optimizations
|  LDFLAGS  = -Wl,--gc-sections -Wl,-rpath,/lib -Wl,-rpath,/usr/local/lib -Wl,-rpath,/usr/lib -L/home/mert/mert/toolchains/openpli40/mipsel-openpli40-linux-gnu/sysroot/usr/lib
|  LIBS     = -lpthread -ldl -lrt
|  UseFlags = 
| Config:
|  Addons   : WEBIF WEBIF_LIVELOG WEBIF_JQUERY TOUCH HAVE_DVBAPI READ_SDT_CHARSETS IRDETO_GUESSING CS_ANTICASC WITH_DEBUG MODULE_MONITOR WITH_LB CS_CACHEEX CW_CYCLE_CHECK CLOCKFIX WITH_EMU
|  Protocols: CAMD35 CAMD35_TCP NEWCAMD CCCAM CCCSHARE GBOX RADEGAST SCAM SERIAL CONSTCW PANDORA GHTTP
|  Readers  : NAGRA IRDETO CONAX CRYPTOWORKS SECA VIACCESS VIDEOGUARD DRE TONGFANG BULCRYPT GRIFFIN DGCRYPT
|  CardRdrs : PHOENIX INTERNAL SC8IN1 MP35 SMARGO DB2COM STAPI STAPI5 STINGER
|  Compiler : mipsel-openpli40-linux-gnu-gcc (crosstool-NG hg+default-86a8d1d467c8) 4.8.1
|  Config   : build/mipsel-openpli40-linux-gnu/config.mak
|  Binary   : ../archiv/oscam-svn11355-openpli40-webif-oscam-emu-patched
+-------------------------------------------------------------------------------
HOSTCC	webif/pages_gen
GEN	webif/pages.c
GEN	Compressed 435752 template bytes into 176841 bytes. 258911 saved bytes (59.42%).
CC	cscrypt/bn_add.c
CC	cscrypt/bn_asm.c
CC	cscrypt/bn_ctx.c
CC	cscrypt/aes.c
CC	cscrypt/bn_div.c
CC	cscrypt/bn_exp.c
CC	cscrypt/bn_lib.c
CC	cscrypt/bn_print.c
CC	cscrypt/bn_mul.c
CC	cscrypt/bn_shift.c
CC	cscrypt/bn_sqr.c
CC	cscrypt/bn_word.c
CC	cscrypt/mem.c
CC	cscrypt/des.c
CC	cscrypt/i_cbc.c
CC	cscrypt/i_ecb.c
CC	cscrypt/i_skey.c
CC	cscrypt/md5.c
CC	cscrypt/rc6.c
CC	cscrypt/sha1.c
CC	csctapi/atr.c
CC	csctapi/icc_async.c
CC	csctapi/io_serial.c
CC	csctapi/protocol_t0.c
CC	csctapi/protocol_t1.c
CC	csctapi/ifd_db2com.c
CC	csctapi/ifd_mp35.c
CC	csctapi/ifd_phoenix.c
CC	csctapi/ifd_sc8in1.c
CC	csctapi/ifd_sci.c
CC	csctapi/ifd_smargo.c
CC	csctapi/ifd_stinger.c
CC	minilzo/minilzo.c
CC	module-anticasc.c
CC	module-cacheex.c
CC	module-camd35-cacheex.c
CC	module-camd35.c
CC	module-cccam-cacheex.c
CC	module-cccam.c
CC	module-cccshare.c
CC	module-constcw.c
CC	module-emulator.c
CC	module-emulator-osemu.c
CC	module-emulator-stream.c
CC	ffdecsa/ffdecsa.c
CC	module-csp.c
CC	module-cw-cycle-check.c
CC	module-dvbapi-chancache.c
CC	module-dvbapi.c
CC	module-gbox-helper.c
CC	module-gbox-sms.c
CC	module-gbox-cards.c
CC	module-gbox.c
CC	module-ird-guess.c
CC	module-monitor.c
CC	module-newcamd.c
CC	module-newcamd-des.c
CC	module-pandora.c
CC	module-ghttp.c
CC	module-radegast.c
CC	module-scam.c
CC	module-serial.c
CC	module-stat.c
CC	module-webif-lib.c
CC	module-webif-tpl.c
CC	module-webif.c
module-webif.c: In function 'send_oscam_files':
module-webif.c:6421:24: error: 'MNU_CFG_FSOFTCAMKEY' undeclared (first use in this function)
   { "SoftCam.Key",     MNU_CFG_FSOFTCAMKEY,FTYPE_CONFIG },  // id 27
                        ^
module-webif.c:6421:24: note: each undeclared identifier is reported only once for each function it appears in
Makefile:437: recipe for target 'build/mipsel-openpli40-linux-gnu/module-webif.o' failed
make[1]: *** [build/mipsel-openpli40-linux-gnu/module-webif.o] Error 1
make[1]: *** Bitmemiş işler için bekliyor....
Makefile:389: recipe for target 'all' failed
make: *** [all] Error 2

oscam-emu, 2017-02-09:
Yes, I know. Will be fixed in Sunday. Be patient.

sculpneo, 2017-02-09:
thanx :)

joeusercz, 2017-02-10:
[oscam-emu.11355.zip](https://github.com/oscam-emu/oscam-emu/files/766166/oscam-emu.11355.zip)

closed 2017-02-12 by oscam-emu