oscam-emu/oscam-emu #233: vmcam integration [feature-request]

hpmueller1971, 2017-02-17:
Would be cool if you could integrate [vmcam](https://github.com/macm00v/vmcam) (for CAID5601)...

oscam-emu, 2017-02-18:
What are you trying to achieve? Do you want to read your 5601 card? If so, then you should ask the oscam guys to support it.

hpmueller1971, 2017-02-18:
No - vmcam is an emu for IPTV-Encryption, certified boxes have no card, they simply connect to a preconfigured server to get the control-words, authentication is just the MAC of the Amino Box. If it's integrated in your patch, the parameters could be read from the SoftCam.Key like a key, ie

C 192.168.1.1,12345,192.168.1.2,23456,11111,company

just like any other supported CAS... so instead of running oscam+osemu+vmcam (or oscam-emu+vmcam), it would be just oscam-emu...

ua0lnj, 2017-02-28:
For vmcam see http://www.streamboard.tv/oscam/ticket/4588
I use vdr->dvbapi->oscam->vmcam
tvheadend->(dvbapi or newcamd)->oscam->vmcam
With this patch all works

oscam-emu, 2017-10-22:
This is a reader for oscam. Nothing to do with emulation.

closed 2017-10-22 by oscam-emu