oscam-emu/oscam-emu #234: Enhancement code correctly on finding the right BISS key

ceco2000, 2017-03-05:
Hi to all developers
  I would like to ask there is an option to improve the code on the right finding the right biss key. The situation now is such that if they are described more key's to / 00011FFF / is obtained mess and the channel remains not decoded even when described the correct key.

Thank you in advance!

oscam-emu, 2017-03-05:
Hi, you can always use the constant cw reader (built in oscam) to describe in detail each channel.

ceco2000, 2017-03-05:
Hi
For me, your answer is not the solution, I think oscam-emu could replace constant.cw module completely.
However, many thanks.

oscam-emu, 2017-03-08:
constant cw protocol is built is in oscam. Why do you ask from an add-on (oscam-emu) for a feature that is already provided without the add-on?

Why can't you use the constant cw protocol?

ceco2000, 2017-03-08:
The explanation and very simple, work and care only one key file.

closed 2017-03-25 by oscam-emu