oscam-emu/oscam-emu #236: PowerVU EMM key capacity

badproject, 2017-04-12:
My SoftCam.Key have more than 100 EMM Keys. I see all keys at reader entitlements.
But oscam-emu use only first 64 EMM keys when autoroll.

oscam-emu, 2017-04-26:
Hi, It seems there's no limitation in the code. It might be a driver issue. 100 keys are way too many.

Try to give a log of oscam-emu with debug level 192 (64=emm + 128=dvbapi).  It might indicate what's wrong.

badproject, 2017-04-26:
File attached.

[oscam.log.zip](https://github.com/oscam-emu/oscam-emu/files/959721/oscam.log.zip)

PeterMaxu, 2018-11-01:
Problem still exist in the actual version, tested with openpli 6.2 on vu solo 4k.

nautilus7, 2018-11-07:
I believe I found the problem! Would you like a test build to see if it works? What is your box (image and cpu architecture)?

oscam-emu, 2018-11-07:
@badproject , @PeterMaxu I believe we found the problem! Would you like a test build to see if it works? What is your box (image and cpu architecture)?

oscam-emu, 2018-11-07:
https://github.com/oscam-emu/oscam-emu/issues/286

oscam-emu, 2018-11-07:
I am not sure how you came into the number of 64 emm keys max. My calculations say that there is a limit of 16 emm keys per group id.

While 16 keys are not that low, I am thinking of increasing it to 32 emm keys per group id.

oscam-emu, 2018-11-07:
With emu r776 emm key capacity was increased to 32 per group id. It should be enough.

closed 2018-11-07 by oscam-emu