oscam-emu/oscam-emu #238: stream-relay_enable wrong default !

mihaip1948, 2017-05-31:
stream_relay_enable  is not initialized and interpreted by
Oscam in execution as "enabled" !!!

oscam-emu, 2017-05-31:
Can you give a more detailed explanation?

mihaip1948, 2017-05-31:
If I have no stream_relay_enable variable initialized into
oscam.conf .... and start Oscam EMU ... in WebIF you
can find : Stream relay ..."enabled" !!! this mean the
default value is either not set ... or have from the
beggining a wrong value ! now if I restart Oscam and
examine oscam.conf I will find stream_relay_enabled=1 !!!

oscam-emu, 2017-05-31:
I cannot reproduce.

If I delete the stream relay related lines from oscam.conf and restart (many times), nothing is added back to oscam.conf.

What you see in webif doesn't really matter unless you actually press SAVE in the stream relay tab in webif. Then "stream_relay_enabled" option is added back to oscam.conf.

Obviously, when the "stream_relay_enabled" variable is not set in oscam.conf, webif default to "1- enabled", but it isn't enabled actually, until you press SAVE, as I said before.

closed 2017-07-21 by oscam-emu