oscam-emu/oscam-emu #240: Tandberg improvement fro same entitlement channels

zinonino, 2017-07-14:
I see different channels with same 'entitlement' and different keys and can use only one, must be good to have options like PowerVU unique groupid to work more channels with the same entitlement and different keys.

oscam-emu, 2017-07-21:
Same entitlement id with different ECM key is very rare in Tandberg. Do you have any example channels with same entitlement and different ecms keys?

It looks like to much work for too little benefit to implement something like this.

zinonino, 2017-07-21:
Yes,

Hellas SAT 39°E
TP 11135,V,30000,8PSK
T 000001 01 5FDXXXXXXXXXXXXX
T 000002 01 CA1XXXXXXXXXXXXX
T 03 01 526XXXXXXXXXXXXX ; Arena Sport 3
T 04 01 345XXXXXXXXXXXXX ;Arena Sport 4
T 05 01 9EFXXXXXXXXXXXXX ; Arena Sport BIH

T 1 01 CA1XXXXXXXXXXXXX - Service 1
T 2 01 5FDXXXXXXXXXXXXX - Service 2
T 3 01 345XXXXXXXXXXXXX - Service 3
T 4 01 526XXXXXXXXXXXXX - Service 4
T 5 01 E4AXXXXXXXXXXXXX - Service 5

EPL Muxx @ 10E 11304 H 30000

http://www.sat-universe.com/showpost.php?p=2036745372&postcount=358
http://www.sat-universe.com/showthread.php?t=295823

Use same ID ,but different ECM key

oscam-emu, 2017-07-23:
As I said it's extremely rare and these channels are not regular channels, but feeds. If the issue becomes more common I will think adding a fix.

closed 2017-07-23 by oscam-emu