oscam-emu/oscam-emu #241: compile warnings

cb3hn, 2017-10-13:
`
CC	module-webif-tpl.c
module-emulator-osemu.c: In function ‘PowervuCalculateCw’:
module-emulator-osemu.c:3463:22: warning: ‘val’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  uint8_t seedLength, val;
                      ^
CC	webif/pages.c
CC	module-webif.c
CC	reader-common.c
module-emulator-osemu.c: In function ‘PowervuECM’:
module-emulator-osemu.c:3564:10: warning: ‘hashModeCw’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  uint8_t hashModeCw, needsUnmasking, xorMode;
          ^
`

How can I fix that?

oscam-emu, 2017-10-13:
This is not a problem actually. You can ignore it. Depending on the compiler might or might not this warning.

Maybe, I'll fix in the future.

oscam-emu, 2017-10-24:
Fixed with r753

closed 2017-10-24 by oscam-emu