oscam-emu/oscam-emu #243: Crash with DREcrypt ee keys

Unb9rn, 2017-10-14:
Tried to make it work with ee keys-oscam just crashes without any error. Trying to run it on Vu+ Duo 2 (mipsel)
I've tried some pre-built binaries, tried to build one myself-all behave the same way. In fact, the only binary working fine is "Oscam r11310 emu r733" built by megavolt85  from one of russian forums. 

In case logs with max verbosity are helpful in any way:
[Here](https://pastebin.com/4UwccuX0)

oscam-emu, 2017-10-14:
I don't have reception on DREcrypt channels, so I can't really help. You'll have to wait for someone else to help.

loxel, 2017-10-14:
Just add to prov # 4AE1 about oscam starts to restart and hangs

label                         = emulator
protocol                    = emu
device                       = emulator
caid                          = 4AE1
detect                       = cd
ident                         =  4AE1:000011
group                        = 1
emmcache                = 1,1,2,0
extee36                     = /etc/tuxbox/config/oscam-emu/ee.bin

When the 4AE1 is reset, the oscam normally operates.
Simply typing 4AE1 causes problems.

Unb9rn, 2017-10-14:
As for me, oscam stops completely when I change into dre-crypt channel. While on FTA, problematic binary works, but as soon as I switch channel to encrypted-it closes. (checked via ps -ax. the process completely disappears.)
About debugging. If you need SSH access to the receiver or http stream with ECM without descrambling-I am willing to provide you with that.
Thanks!

oscam-emu, 2017-11-07:
@Unb9rn contact us via email (the email address is at the bottom of the main page) to talk about the issue.

oscam-emu, 2017-11-14:
fixed with r756

closed 2017-11-14 by oscam-emu