oscam-emu/oscam-emu #244: MultiPorts

fairbird, 2017-10-27:
Why Oscam doesn't supported MultiPorts in one reader ?!
As  like as wicarrd or Mgcamd ..
Thank you

oscam-emu, 2017-10-27:
I have no idea what "multiports" is good for and I don't give a damn what wicard does...

If you want, you can try to describe what are you trying to achieve with this "multiports" feature. Then we maybe add this functionality to our emu.

Else you can ask the developers of wicard to provide patches with the functionality you want. Instead of just stealing our code, they can for once give some code back...

fairbird, 2017-10-27:
Thank you for answer ..
I've means about "multiports" such as these lines for newcamd server (CCcam emu also can support it)

CWS = 46.4.158.144 31200 6kj470 RRR 01 02 03 04 05 06 07 08 09 10 11 12 13 14
CWS = 46.4.158.144 21000 6kj470 RRR 01 02 03 04 05 06 07 08 09 10 11 12 13 14
CWS = 46.4.158.144 21311 6kj470 RRR 01 02 03 04 05 06 07 08 09 10 11 12 13 14
CWS = 46.4.158.144 21113 6kj470 RRR 01 02 03 04 05 06 07 08 09 10 11 12 13 14

and should be like this in reader same as wicardd

[reader]
active = 1
name = newcamd
type = newcamd
filter = dvb,bloker
auto_provider = all
auto_caid = 1
auto_idents = 1
auto_update = 1
account = 6kj470:RRR@46.4.158.144:0102030405060708091011121314
**multiport**=31200.21000.21311.21113
debug = 1
reconnect_delay = 1
emm_cache = 1

oscam-emu, 2017-10-27:
This has nothing to do with emu. It's related to oscam and its readers. Ask there for supporting "multiports".

closed 2017-10-27 by oscam-emu