oscam-emu/oscam-emu #246: about adding optional pid limiter

egydoctor2010, 2017-11-03:
hi all,
can we include in oscam patch pid limiting as joeuser's oscam version 11360 and 11384 to be optional
as sh4 stb can not open powervu channels unless joeuser's drivers are added to enigma image
but when stream pids of the opened powervu channel are more than 5, stb freeze and need to be rebooted either manually or by reboot command in telnet 
and not accepting killall command.
i want to make these changes optional only for users just like me or any one has sh4 stb
here is a screen shot for joeuser's version
[image]

budiarno, 2017-11-03:
This is not related to OSCam-Emu, try using patch by Joe (forgot where he posted the file, may be on Streamboard or Linuxsat-Support).

egydoctor2010, 2017-11-03:
joe did not make patch he edited 2 versions as a whole but i ask if these  changes can be added

joeusercz, 2017-11-08:
I vaguely remember posting a patch, but I do not remember where... :)
I am not sure if it should be included in oscam and/or oscam-emu. If it is configured wrong, it could cause very hard to trace problems for users.
Now that VU+ boxes support powervu, it might be useful for those boxes??? Will have to ask around (I do not have any VU+ boxes.)
I will try to update the source code on my github account when I get a chance...

joeusercz, 2017-11-08:
here is a diff...

joeusercz, 2017-11-08:
Here is a diff...
[oscam_pid_limiter.11384.diff.txt](https://github.com/oscam-emu/oscam-emu/files/1455261/oscam_pid_limiter.11384.diff.txt)

oscam-emu, 2017-11-09:
@joeusercz,

Is this pid limiting setting only useful when stream relay is used? Or this can be used in other cases as well?

I ask because I am willing to make a request to the oscam guys to accept the patch. Do I have any chances? Otherwise, I'll included in the Emu.

egydoctor2010, 2017-11-10:
@oscam-emu 
this is pid limiting working with our drivers in sh4 devices without stream relay. we did not try it with stream relay but it is very helpful for us please include it in the emu if you please for seek of our sh4 devices
as when i watch for example any powervu channels in afn 9e the stb freeze must be rebooted as these channels has about 10 to 12 stream pids , but when watching channels like betfred in 4.8e it works ok 
so when i want to watch afn i use joeuser's oscam 11384 and after finished i return to the new oscam 11392 so please include it.

BarviX6, 2017-11-10:
https://github.com/joeusercz
drivers HW DES Taapat image
1 work 4 (maybe more) audio tracks
2 works only with oscam (2--8)-pids
3 oscam (64)-pids -------- freezes the box

-----------
drivers HW DES HDMU/ATV image
1 work 3 audio tracks
2 works with oscam (64)-pids

oscam-emu, 2017-11-14:
added with r757

closed 2017-11-14 by oscam-emu