oscam-emu/oscam-emu #247: Please Add this ..

fairbird, 2017-11-06:
https://github.com/Hains/oscam/commit/f423419f19c2ec2aa7f2b84cfe455510a5196521

oscam-emu, 2017-11-06:
This is a pure oscam modification. Nothing to do with the emu.
I believe you should explain to the oscam guys what it does and why this is needed and ask them to commit it in the oscam svn. 

Otherwise, you can tell me what it does and maybe I make a pull request to oscam.

fairbird, 2017-11-06:
I have add it in my emu build tree and test it ..
Before with some new image as like as ARM boxes emu doesn't work..
And now work just fine ..

Because those image have newest version of openssl..

oscam-emu, 2017-11-09:
@fairbird as I explained before, these 2 requests are not related to the emu. They are related to oscam. Make a patch and ask the oscam guys to accept it.

closed 2017-12-30 by oscam-emu