oscam-emu/oscam-emu #248: videoguard v4

spein46, 2017-11-21:
hi, sorry, I open this isse but I hope to find some attention.
I would like study to project an oscam emu of new videoguard system, but I don t find anyone that konow about this.
is there anyone interested?
thanks

spein46, 2017-11-21:
thanks for the reply.
I would read my sky it card in enigma box, but there isn't emu to read new sky it card

oscam-emu, 2017-11-21:
Still not clear what you're trying to achieve...

As far as I know Sky Italia is paired which means the card can only be read on the Sky box.

spein46, 2017-11-21:
yes, it is.
oscam emu can emulate seca irdeto, videoguard v1, v2, v3. I read this.
the paired of sky italia is videoguard upgrade or not?
I would like build oscam emu aboout this new system of sky ita, is it possible?

oscam-emu, 2017-11-21:
It seems that you are a little bit confused...

closed 2017-11-21 by oscam-emu

spein46, 2017-11-21:
why?can you help me to understand better?

oscam-emu, 2017-11-22:
Yes. This is what pairing is supposed to do. The card only works in that specific receiver that Sky gave you. No other receiver, even if it is exactly the same model.