oscam-emu/oscam-emu #250: some freeze at biss channel name TV-Varzish

pop-dahaka, 2017-11-28:
Hi all team ,
sat 52.5E 
11785 H 27500 8PSK 
Spark 7111 
I tried both constant.cw and SoftCam.Key but both freeze and cause unwatchable (video and audio) , but other biss channels works so good and the problem only with this channel .
Testing mcas found no freeze , but i like oscam and wish you fix this problem soon .

oscam.conf
`[global]
logfile                       = stdout
pidfile                       = /var/run/oscam.pid
nice                          = -1
maxlogsize                    = 1000
disablecrccws_only_for        = 0500:032830,030B00;1811:003311;0100:003311

[streamrelay]
stream_relay_enabled          = 0
stream_emm_enabled            = 0

[dvbapi]
enabled                       = 1
au                            = 1
delayer                       = 60
user                          = DM800hdse
read_sdt                      = 2
write_sdt_prov                = 1
extended_cw_api               = 0
boxtype                       = dreambox`

oscam.server
`[reader]
label                         = emulator
protocol                      = emu
device                        = emulator
caid                          = 0500,0604,0648,0650,090F,0D00,0D02,0D03,0D05,0D95,0D98,0E00,0F00,1010,1801,1FFF,2600,4AE1,FFFF
detect                        = cd
ident                         = 0500:000000,030B00,023800,021110,007400,007800;0604:000000;0648:000000;0650:000000;090F:000000;0D00:000000,000004,000010,000014,000020,0000C0,0000C4,0000C8,0000CC;0D02:000000,00008C,0000A0,0000A4,0000A8;0D03:000000,000004,000008,000024,000028;0D05:000000,000004,000010;0D95:000000;0D98:000000;0E00:000000,000002;1010:000000;1801:000000,007301,001101;2600:000000;4AE1:000000,000011,000014,0000FE;FFFF:000000
group                         = 1,2,3,4
emmcache                      = 1,1,2,1
auprovid                      = 0500:030B00;0604:010200;0E00:000000;1010:000000;4AE1:000000

[reader]
label = emutest
protocol = constcw
device = /var/keys/constant.cw
caid = 2600
group = 5`

Thanks for your good work and best wishes .

oscam-emu, 2017-11-29:
BISS channels don't freeze. They either clear (if you have the correct key), or they don't clear and you get a black screen.

Since constant cw also gives you problems, maybe you have a bad oscam binary, built for another architecture or something.

closed 2017-11-29 by oscam-emu

pop-dahaka, 2017-11-29:
Its clear and key is correct , 
I use OScam emu by both hapeba and juli13 , here you are two logs for both :

OSCam-EMU (current Revision: HPB r741 - SB r11392 - EMU r417-754)
build_dir_sh_4 = Octagon, Ipbox, Kathrein, Fortis, Spark-Box

**LOG HPB**
key at SoftCam.key
`

    Status Live Log Config Readers Users Services Files Failban CacheEX	Scripts Shutdown/Restart 

BOTTOM

    29.11.2017 10:24:47 00000000 s     (main) System name    = Linux
    29.11.2017 10:24:47 00000000 s     (main) Host name      = stlinux
    29.11.2017 10:24:47 00000000 s     (main) Release        = 2.6.32.42_stm24_V4.1-SAT7111_7111-STSDK
    29.11.2017 10:24:47 00000000 s     (main) Version        = #56 PREEMPT Tue Oct 16 16:17:14 CST 2012
    29.11.2017 10:24:47 00000000 s     (main) Machine        = sh4
    29.11.2017 10:24:47 00000000 s     (main) creating pidfile /var/run/oscam.pid with pid 2571
    29.11.2017 10:24:47 00000000 s      (emu) OSCam-Emu version 754
    29.11.2017 10:24:47 00000000 s     (main) signal handling initialized
    29.11.2017 10:24:47 00000000 s   (config) 9 provid's loaded
    29.11.2017 10:24:47 00000000 s   (config) 415 service-id's loaded in 11 ms
    29.11.2017 10:24:47 00000000 s   (reader) emulator [emu] creating thread for device emulator
    29.11.2017 10:24:47 00000000 s   (reader) emutest [constcw] creating thread for device /var/keys/constant.cw
    29.11.2017 10:24:47 00000000 s     (main) waiting for local card init
    29.11.2017 10:24:47 68E80D1E p  (constcw) Local reader: emutest (file: /var/keys/constant.cw)
    29.11.2017 10:24:47 00000000        (emu) INFO: FFDecsa parallel mode = 32
    29.11.2017 10:24:47 68E80D1E p   (reader) emutest [constcw] proxy initialized, server /var/keys/constant.cw:0
    29.11.2017 10:24:47 60A7DAC2 r      (emu) Reading key file: /var/keys/SoftCam.Key
    29.11.2017 10:24:47 50297343 h    (webif) webif: decompressed 211530 bytes back into 494688 bytes
    29.11.2017 10:24:47 50297343 h    (webif) HTTP Server running. ip=0.0.0.0 port=8888
    29.11.2017 10:24:47 60A7DAC2 r   (reader) emulator [emu] Reader initialized (device=emulator, detect=cd, mhz=357, cardmhz=357)
    29.11.2017 10:24:48 60A7DAC2 r   (reader) emulator [emu] card detected
    29.11.2017 10:24:49 60A7DAC2 r   (reader) emulator [emu] found card system emu
    29.11.2017 10:24:49 60A7DAC2 r   (reader) emulator [emu] THIS WAS A SUCCESSFUL START ATTEMPT No  1 out of max alloted of 1
    29.11.2017 10:24:49 60A7DAC2 r      (emu) Reading key file: /var/keys/SoftCam.Key
    29.11.2017 10:24:50 00000000 s     (main) init for all local cards done
    29.11.2017 10:24:50 00000000 s (emmcache) loading emmcache disabled
    29.11.2017 10:24:50 00000000 s (emmcache) loading emmstat disabled
    29.11.2017 10:24:50 00000000 s (anticasc) anti cascading disabled
    29.11.2017 10:24:50 57C8EA85 c   (client) plain dvbapi-client granted (anonymous, au=off)
    29.11.2017 10:24:50 57C8EA85 c   (dvbapi) loading channelcache disabled
    29.11.2017 10:24:50 57C8EA85 c   (dvbapi) Detected /dev/dvb/adapter0/demux0 Api: 0, userconfig boxtype: 1 maximum amount of possible filters is 32 (oscam limit is 32)
    29.11.2017 10:24:57 57C8EA85 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 2600 ECM_PID: 1FFF PROVID: 000000 
    29.11.2017 10:24:57 57C8EA85 c   (dvbapi) Demuxer 0 found 1 ECMpids and 2 STREAMpids in caPMT
    29.11.2017 10:24:57 57C8EA85 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 2600 PROVID 000000 ECMPID 1FFF ANY CHID PMTPID 002E VPID 1011
    29.11.2017 10:24:57 68E80D1E p  (constcw) Searching CW for CAID 2600 PROVID 000000 SRVID 0005 ECMPID 1FFF PMTPID 002E VPID 1011
    29.11.2017 10:24:57 60A7DAC2 r      (emu) Key found: F 5A00CE68 00
    29.11.2017 10:24:57 57C8EA85 c      (ecm) anonymous (2600@000000/0000/0005/13:C8ADE79909235474F0FAB0624BE1C1E8): found (35 ms) by emulator (P/2/2/2) - TV-Varzish
    29.11.2017 10:24:58 57C8EA85 c   (dvbapi) Demuxer 0 found 1 ECMpids and 2 STREAMpids in PMT
    29.11.2017 10:26:11 50297343 h    (webif) unauthorized access from 192.168.1.2 - invalid credentials

`

oscam-svn11391-sh_4-webif-emu
**LOG juli13**
key at SoftCam.key
`

    2017/11/29 10:36:20 00000000 s     (main) System name    = Linux
    2017/11/29 10:36:20 00000000 s     (main) Host name      = stlinux
    2017/11/29 10:36:20 00000000 s     (main) Release        = 2.6.32.42_stm24_V4.1-SAT7111_7111-STSDK
    2017/11/29 10:36:20 00000000 s     (main) Version        = #56 PREEMPT Tue Oct 16 16:17:14 CST 2012
    2017/11/29 10:36:20 00000000 s     (main) Machine        = sh4
    2017/11/29 10:36:20 00000000 s     (main) creating pidfile /var/run/oscam.pid with pid 3848
    2017/11/29 10:36:20 00000000 s      (emu) OSCam-Emu version 754
    2017/11/29 10:36:20 00000000 s     (main) signal handling initialized
    2017/11/29 10:36:20 00000000 s   (config) 9 provid's loaded
[... 65 lines cut ...]
    2017/11/29 10:41:45 76E59098 c   (client) plain dvbapi-client granted (anonymous, au=off)
    2017/11/29 10:41:45 76E59098 c   (dvbapi) dvbapi channelcache loaded from /var/keys/oscam.ccache
    2017/11/29 10:41:45 76E59098 c   (dvbapi) Detected /dev/dvb/adapter0/demux0 Api: 0, userconfig boxtype: 1 maximum amount of possible filters is 32 (oscam limit is 32)
    2017/11/29 10:41:45 76E59098 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 2600 ECM_PID: 1FFF PROVID: 000000 
    2017/11/29 10:41:45 76E59098 c   (dvbapi) Demuxer 0 found 1 ECMpids and 2 STREAMpids in caPMT
[... cut ...]
    2017/11/29 10:41:45 529BE078 r      (emu) Key not found: F A11FEED5 00
    2017/11/29 10:41:45 529BE078 r      (emu) ECM failed: Key not found
    2017/11/29 10:41:45 42D5701F p  (constcw) Entry found: 2600@000000:0005:002E:1FFF:1011:12 34 00 46 AB CD 00 78 12 34 00 46 AB CD 00 78 
    2017/11/29 10:41:45 76E59098 c      (ecm) anonymous (2600@000000/0000/0005/13:C8ADE79909235474F0FAB0624BE1C1E8): found (90 ms) by emutest (P/2/2/2) - TV-Varzish

`

Please help me to fix this problem , i can record one minute by mobile for you if you wish .

pop-dahaka, 2017-12-08:
Tested both oscam with emu and without emu all versions freeze at biss channel TV-Varzish ,
The only oscam-ymod NO freeze , now am using oscam-ymod for watching this channel TV-Varzish .

Best Wishes .