oscam-emu/oscam-emu #251: If enabled AU - oscam is crashed

X-Filess, 2017-12-06:
Hello.
If enable AU, oscan is crashed.
I add logs and traces logs.
Please check it..
[crash.txt](https://github.com/oscam-emu/oscam-emu/files/1536909/crash.txt)
[oscam.txt](https://github.com/oscam-emu/oscam-emu/files/1536910/oscam.txt)

oscam-emu, 2017-12-06:
Does it crash every time?
Does it crash with r761 also?
Does it crash only with AU enabled?
Does it crash with local ecms i.e. without using newcamd?

X-Filess, 2017-12-06:
>Does it crash every time?
Yes, if connect connect to newcamd where enabled AU i get crash
>Does it crash with r761 also?
I tested on current version, i not check r761
>Does it crash only with AU enabled?
Yes, if dissable AU, all is ok
>Does it crash with local ecms i.e. without using newcamd?
nope, crash before connecting user where enabled AU

oscam-emu, 2017-12-07:
What channel is it and what satellite?

Can you try without newcamd? Probably now you have oscam-emu running on one box, you watch the channel with another box and you connect the 2 boxes through newcamd. Can you test using the same box, i.e for watching and for running oscam-emu?

Also please try previous versions, as many as you can, so I can determine which change causes the problem.

oscam-emu, 2017-12-07:
Now that I think of it seems it's not related to any change I did, but it is a general problem of newcamd. See issue #219 . It seems the exact same problem, but with tandberg caid instead of powervu.

oscam-emu, 2017-12-07:
#219

X-Filess, 2017-12-07:
-What channel is it and what satellite?
4.8°E | Astra 4A | Europe BSS51 | 12360 V | DVB-S28PSKMPEG-4 | 275005/6 | PowerVu
Channel 2900 - TLC Pan Regional

-Can you try without newcamd?
No, can't.

joeusercz, 2017-12-07:
newcamd does not support powervu. But still the crash should be fixed.
What client are you using? I cannot replicate using Tvheadend as a client.

X-Filess, 2017-12-07:
@joeusercz 
you can try : https://cesbo.com/en/astra/