oscam-emu/oscam-emu #254: Slow channelchange on AFN powervu

stemkow, 2017-12-24:
With oscam-emu Emu r764 - Add PowerVu hash mode 4 and oscam  http://www.streamboard.tv/svn/oscam@11392 channelchange is very slow (10 seconds) on afn powervu.

I have tracked the problem down to module-dvbapi.c line 4454. Channel changing speed is greatly improved  when removing " || pvu_skip". I am not sure if this is a good fix and it may cause problems on other powervu channels. However, it works for me and it helps pinpoint the problem

oscam-emu, 2017-12-25:
Which box and image? Have you tried increasing the delayer option in dvbapi to see if this helps?

Can you post a livelog with and without the pvu_skip in the code?

I don't have this problem. My box zaps normally. In addition, this "pvu_skip" is in the code for a very long time and none has ever said anything about slow zapping.

wphilips, 2017-12-29:
This with latest tvheadend. I use oscam.net protocol >=10389.  CW mode: standard/auto
oscam.delayer was set to 60 ms. 

The two log files are attached. This is from an experiment where I tune to one afn 
channel. As soon as there is picture I tune to a second. 

[good.txt](https://github.com/oscam-emu/oscam-emu/files/1593800/good.txt)
[bad.txt](https://github.com/oscam-emu/oscam-emu/files/1593801/bad.txt)

oscam-emu, 2018-02-13:
I would suggest to ask the tvheadend developers.