oscam-emu/oscam-emu #267: PowerVu 4180V30000 SAT AsiaSat 7 105.5E

arespengmj, 2018-04-18:
now this tp powervu keys auto roll,who have UID and EMM key?please tell me,tks.

budiarno, 2018-04-18:
It wasn't roll up, it's just an update to the PowerVu algo.

intersect23, 2018-04-18:
this tp autoroll with uid. 
send they now emm keys or no - i dont know. 
have only working ecm keys for test.
but can save (if need) stream for analyze 
dvbsnoop -adapter 4 -n 30000 0x005a -b > /tmp/emm.bin

budiarno, 2018-04-18:
[image]
It still uses the same old ECM key (take a look at the cw is still there), but SPN have updated its encryption.

PS: ignore my background channel.

arespengmj, 2018-04-23:
@nrosier @nautilus7 please tell me UID (unique id)&EMM key.ecm key
thank you

intersect23, 2018-04-23:
my keys
P xxxx 00 9BA0B0D45F0A93
P xxxx 01 785E6AFF44D7A9
keys ok, but black screen
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:EA05B7657D17733CDC0150E33BABE7D4): found (0 ms) by OSEmu2
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:7A780BB222DD92B14E84AA24389C0847): found (0 ms) by OSEmu2
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:EA05B7657D17733CDC0150E33BABE7D4): cache1 (0 ms) by OSEmu2
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:7A780BB222DD92B14E84AA24389C0847): cache1 (0 ms) by OSEmu2
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:EA05B7657D17733CDC0150E33BABE7D4): cache1 (0 ms) by OSEmu2
2018/04/23 14:13:52 193F15E7 c      (ecm) cam5 (0E00@000000/0000/0016/7E:7A780BB222DD92B14E84AA24389C0847): cache1 (0 ms) by OSEmu2

also screenshot with some data from one forum
[image]

oscam-emu, 2018-04-23:
@intersect23 are you completely retarded? Seriously, you post keys here?

closed 2018-04-23 by oscam-emu