oscam-emu/oscam-emu #268: Counter attack from the 55.5W sat now at 65 and 75W

scolanzi, 2018-04-25:
Sorry for the new post with the same theme but now the problem has spread to the 65W and 75W satellites. Oscam does the AU, finds ECMkey and in the log informs the channel opening but continues black screen.

scolanzi, 2018-04-25:
Below is the oscam-emu log for a 75w sat channel. Note that after finding ECMkey an EMM error message appears. Is this correct?

2018/04/25 16:21:31 32B97E90 r      (emu) Key not found: P 00A7 01
2018/04/25 16:21:31 32B97E90 r      (emu) ECM failed: Key not found
2018/04/25 16:21:31 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:38040812F5EB87A59002127CCBD18A64): not found (2 ms) by emulator - Viva HD
2018/04/25 16:21:31 1607BBFF c   (reader) emulator [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2018/04/25 16:21:31 1607BBFF c   (reader) emulator [emu] EMM: reader_caid 0000 != emmpid caid 0E01 -> SKIP!
2018/04/25 16:21:31 1607BBFF c   (reader) SoftCam.Key [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2018/04/25 16:21:31 1607BBFF c   (reader) SoftCam.Key [emu] EMM: reader_caid 0000 != emmpid caid 0E01 -> SKIP!
2018/04/25 16:21:31 1607BBFF c   (dvbapi) Demuxer 0 restarting decodingrequests after 0 ms with 1 enabled and 0 disabled ecmpids!
2018/04/25 16:21:31 1607BBFF c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1777 ANY CHID PMTPID 142F VPID 029E
2018/04/25 16:21:32 32B97E90 r      (emu) Key not found: P 00A7 01
2018/04/25 16:21:32 32B97E90 r      (emu) ECM failed: Key not found
2018/04/25 16:21:32 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:55405A0B24E92E21594C62798B97098E): not found (2 ms) by emulator - Viva HD
2018/04/25 16:21:32 1607BBFF c      (emm) emm:
2018/04/25 16:21:32 1607BBFF c      (emm)   82 30 9B 50 99 01 0E 00 31 00 06 8F 00 60 1E D2 
2018/04/25 16:21:32 1607BBFF c      (emm)   C4 32 03 80 7D 47 93 F7 0A 23 B8 EE CF C4 92 5A 
2018/04/25 16:21:32 1607BBFF c      (emm)   E0 7B 79 E9 FD F1 19 2A 85 30 72 71 F7 9E 80 7D 
2018/04/25 16:21:32 1607BBFF c      (emm)   3E 09 54 91 B7 18 04 77 A9 F7 A5 C3 E8 3B 61 8D 
2018/04/25 16:21:32 1607BBFF c      (emm)   A0 DE 47 F7 B7 36 D7 D2 D1 80 7D 6B E2 02 92 A3 
2018/04/25 16:21:32 1607BBFF c      (emm)   06 28 A6 86 9D 15 98 F7 13 75 EA 97 12 72 79 98 
2018/04/25 16:21:32 1607BBFF c      (emm)   1F AC CF 48 80 7D 3C CF B8 5A 28 09 17 48 70 C2 
2018/04/25 16:21:32 1607BBFF c      (emm)   C7 EB AB 9A B6 0D 76 87 CC FD FD 6B 35 0C B7 80 
2018/04/25 16:21:32 1607BBFF c      (emm)   7D 3B 89 C1 FD 45 C5 3E 1D D4 4F 45 E6 91 36 62 
2018/04/25 16:21:32 1607BBFF c      (emm)   A2 7C 8D 0E 75 69 D7 09 EF 36 DB D2 FA DB 
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] EMM: emmtype unique. Reader serial ################.
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] EMM: emm UA/SA: ################.
2018/04/25 16:21:32 1607BBFF c (emmcache) added emmstat (reader:emulator, count:0)
2018/04/25 16:21:32 1607BBFF c (emmcache)   C6 68 12 FC B1 C1 D6 1F 50 5B E1 87 DE 4F F5 90 
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] EMM: emm count 0 rewrite 2
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] EMM: emm is being sent to reader
2018/04/25 16:21:32 1607BBFF c   (reader) emulator [emu] Successfully added EMM to /etc/tuxbox/config/oscam/emulator_unique_emm.log
2018/04/25 16:21:32 1607BBFF c   (reader) SoftCam.Key [emu] EMM: reader 0E00 match since emmpid has no provid -> SEND!
2018/04/25 16:21:32 1607BBFF c   (reader) SoftCam.Key [emu] EMM: emmtype unique. Reader serial ################.
2018/04/25 16:21:32 1607BBFF c   (reader) SoftCam.Key [emu] EMM: emm UA/SA: ################.
2018/04/25 16:21:32 1607BBFF c (emmcache) added emmstat (reader:SoftCam.Key, count:0)
2018/04/25 16:21:32 1607BBFF c (emmcache)   C6 68 12 FC B1 C1 D6 1F 50 5B E1 87 DE 4F F5 90 
2018/04/25 16:21:32 1607BBFF c   (reader) SoftCam.Key [emu] EMM: emm count 0 rewrite 1
2018/04/25 16:21:32 1607BBFF c   (reader) SoftCam.Key [emu] EMM: emm is being sent to reader
2018/04/25 16:21:32 1607BBFF c   (dvbapi) Demuxer 0 restarting decodingrequests after 0 ms with 1 enabled and 0 disabled ecmpids!
2018/04/25 16:21:32 1607BBFF c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1777 ANY CHID PMTPID 142F VPID 029E
2018/04/25 16:21:32 716DCA0D r (emmcache) added emm to cache:
2018/04/25 16:21:32 716DCA0D r (emmcache)   C6 68 12 FC B1 C1 D6 1F 50 5B E1 87 DE 4F F5 90 
2018/04/25 16:21:32 32B97E90 r (emmcache) found emmstat match (reader:emulator, count:0)
2018/04/25 16:21:32 32B97E90 r (emmcache)   C6 68 12 FC B1 C1 D6 1F 50 5B E1 87 DE 4F F5 90 
2018/04/25 16:21:32 716DCA0D r (emmcache) found emmstat match (reader:SoftCam.Key, count:0)
2018/04/25 16:21:32 716DCA0D r (emmcache)   C6 68 12 FC B1 C1 D6 1F 50 5B E1 87 DE 4F F5 90 
2018/04/25 16:21:32 32B97E90 r      (emu) Key found in EMM: P 0005**** 00 53861AE4F26CC6 ; UA: 00601ED2
2018/04/25 16:21:32 32B97E90 r        (-) -- Skipped 1 duplicated log lines --
2018/04/25 16:21:32 32B97E90 r      (emu) Writing key file: /etc/tuxbox/config/oscam/SoftCam.Key
2018/04/25 16:21:32 32B97E90 r      (emu) Key written: P 500A1 01 E6F407B3B370FA
2018/04/25 16:21:32 716DCA0D r      (emu) Writing key file: /etc/tuxbox/config/oscam/SoftCam.Key
2018/04/25 16:21:32 716DCA0D r      (emu) Key written: P 500A2 01 E6F407B3B370FA
2018/04/25 16:21:32 32B97E90 r      (emu) Writing key file: /etc/tuxbox/config/oscam/SoftCam.Key
2018/04/25 16:21:32 32B97E90 r      (emu) Key written: P 500A3 01 E6F407B3B370FA
2018/04/25 16:21:32 1607BBFF c      (emm) emm:
[... cut ...]
2018/04/25 16:21:33 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:2B367BE03B85958021FDBF4E4E46E0C8): found (4 ms) by emulator - Viva HD
2018/04/25 16:21:34 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:BB090AB82D2BD5347A13060BAF788D4C): found (4 ms) by emulator - Viva HD
2018/04/25 16:21:35 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:E3ABCB2C9C1D8EEC0BFF6D3452FD73A8): found (4 ms) by emulator - Viva HD
2018/04/25 16:21:36 1607BBFF c      (ecm) dvbapi (0E00@000000/0000/00A7/48:C5F5D4A97B

ermanske, 2018-06-06:
Espero que ajude, uma gravação fox sports 40.5W ...

ECM[Video]: 
ECM: 80 30 45 50 3F 20 0E 00 10 08 F2 4B 33 22 F4 96 58 4B 00 06 B0 00 01 EB C9 01 A1 52 A1 E1 32 52 BE 75 16 CD B4 17 20 00 00 97 B7 E4 88 EB 8F C0 94 E0 E8 D8 1A E1 C8 B0 0C 17 59 25 F4 BA 62 4C 42 23 E3 59 0B 09 57 CE
Unmasked ECM: 80 30 45 30 3F 20 0E 00 00 08 F2 4B 33 22 F4 96 58 4B 00 06 B0 00 01 EB C9 01 60 5A A1 E1 32 52 BE 75 16 EB 00 17 20 00 00 97 E8 47 CC 2F 8F C0 97 E0 29 48 89 E1 C8 B0 0C 17 45 25 A8 83 62 4C 42 16 8C 59 70 DC 1E 70
IsDecrypted: True
DVB Scrambling Algo: DES
Needs Unmasking: True
ECM Counter: $06
Nanos: [5]->($20, $3E) 
Nano $20 Extra Data Length: 8
Hash Mode CW: 1, ECM: 10
Seed CW ECM: 01 A1 01 52 EB BE C9 E1 16 32 75 00 00 00 00 00 
ECM CalculateSeed: 80 30 45 30 3F 20 0E 00 00 08 F2 4B 33 22 F4 96 58 4B 00 06 B0 00 B9 F8 A7 74 0F B0 0C 41 C0 14 39 EE 4E 0C 00 17 20 00 00 2C 00 A9 B9 B6 01 01 85 73 69 75 66 02 02 A0 82 50 00 00 00 A6 00 00 00 BB 17 20 70 DC 1E 70 
Base Cw: 0C 41 C0 14 39 EE 4E 
Seed VPid: 9B 98 C1 A9 6A 58 00 00 
Seed APid-1: 35 26 4A 60 98 27 00 00 
Seed APid-2: 35 26 4A 60 98 27 00 00 
Seed APid-3: B7 E0 A8 44 11 29 00 00 
Seed APid-4: B7 E0 A8 44 11 29 00 00 
Seed HSD: 79 90 E2 6C 9B 2E 00 00 
S-Box Type: Primary Tables
Xor Mode: Normal mode[0]
Ident: $1720
KeyIdx: 1
Key: C2 3B B4 69 95 A1 C9 
OddCwAudio1: 2C A8 51 02 D5 10 13 6D 
EvenCwAudio1: 80 EF 4C 2F F7 67 FE C2 
OddCwAudio2: 2C A8 51 02 D5 10 13 6D 
EvenCwAudio2: 80 EF 4C 2F F7 67 FE C2 
OddCwAudio3: FD C8 94 FB CB EA C1 E3 
EvenCwAudio3: 8C 54 62 6B F2 1A FE 51 
OddCwAudio4: FD C8 94 FB CB EA C1 E3 
EvenCwAudio4: 8C 54 62 6B F2 1A FE 51 
OddCwVideo: 16 76 D3 A4 8C 70 C7 FE 
EvenCwVideo: F8 79 2F 62 02 CD 6B A4 
OddCwHSD: 91 EA D6 8C AD A4 61 02 
EvenCwHSD: D0 F7 7C 02 70 16 B6 38 

TS Before CSA: 
47 0C 8A D5 0C AF 42 5C 05 DD F8 E6 DA 47 B3 D7 38 06 80 A3 01 48 7C 74 CA F5 E6 4D B7 FC 2B 5F E9 E4 3A 1F 3A 2D 17 A0 05 83 8C C2 43 28 D3 45 46 25 E1 8F BB C4 18 9A 72 6C 6F 69 07 5F 11 7B E2 39 48 C6 F9 F7 A7 A4 B5 26 C1 41 EA CF 76 57 6F 65 A1 26 59 6A DC 29 94 5D EC 81 C1 5A B3 D1 3C A5 2F 0D BD E1 C6 61 68 EE B7 12 E8 77 E4 FB 3C E4 1F BD 9C B8 87 62 A1 61 3B 98 9C E2 3B B7 58 24 B6 49 63 61 22 A8 C3 E6 08 14 C2 18 37 FB 13 28 B7 1B 87 BE B4 FD E0 B2 60 38 FD B1 D8 5F 02 5D 98 E1 70 85 2A 3B 24 36 A5 A9 7F 9F AC A5 7A F0 E7 1D D8 FC 28 65 A9 CE 59 74 

TS After CSA: 
47 0C 8A 15 CE 3A 82 B7 15 F6 A1 62 D5 B2 66 F4 0D CC 05 4A 98 57 B0 2A B1 D3 75 3B CF C4 87 51 1C A4 05 37 BD 9E E6 22 7E 8E 0A 0C 0F 67 21 7A F9 38 D8 CD 2F D8 9E 1F B8 6A 38 E1 D0 93 6B 82 C5 A5 F3 C7 33 AE 2A 7E 3F 37 0F 33 70 1A 85 37 50 3E D5 BE 36 40 EB B5 CF 5C C3 E0 6F C1 0E 56 44 57 B0 B7 A7 B7 29 60 68 E8 34 9D 71 FF B0 6F A2 EF 73 58 0E CE 8F 7D C5 E3 CA 72 89 A6 06 87 D1 C9 C6 80 57 E1 D7 DF 11 FE 15 15 15 66 EF 7F FC B6 C1 A6 FF DD 17 41 6A AB AF CE 3D ED 89 D4 66 2A 80 5E B1 E5 10 D9 0A EE 94 BE F9 51 CE CE D7 C6 86 3C EF B5 24 48 05 E5 6C 47

Error 
EvenCWvideo
OddcwAudio

Keys:
P 0000 00 5100C379AF73E5 ; 
P 0000 01 C23BB46995A1C9 ;
[image]

oscam-emu, 2018-06-07:
Is this some kind of joke? Why do you keep posting logs and asking for powervu to be fixed? If you know how to fix it, post a patch. If you don't, just stop asking.

closed 2018-06-07 by oscam-emu