oscam-emu/oscam-emu #273: New parameter mux globosat 65W and 75W

jlmariense, 2018-07-02:
Hello friends, just the mux globosat of 65W and 75W satellites, coded. Below the oscam log:
2018/07/02 17:34:40 04D832E1 r      (emu) ecm1:803045503F200E0078089276262863E95F5F00EAB00077734C0CF5DA9DE4F40DE367039C2808400000CD5B493858159C8E1C6F7441B7EEA451411FAE212058CC527E3BDB893B29B4
2018/07/02 17:34:40 04D832E1 r      (emu) csaUsed=0,xorMode=0,ecmSrvid=132,hashModeCw=1,modeCW=4

budiarno, 2018-07-05:
SPN on Asiasat 7 is also using the same modeCW 4.

`csaUsed=1,xorMode=0,ecmSrvid=0,hashModeCW=1,modeCW=4`

oscam-emu, 2018-10-20:
Test with r774 https://github.com/oscam-emu/oscam-emu/commit/d8f4f3d7ff4c4cbe67da014a05ee95322689aba2 if it is fixed

scolanzi, 2018-11-07:
Now hash mode 25