oscam-emu/oscam-emu #278: Oscam crash with corruption (!prev) when use Softcam.key

zinonino, 2018-07-25:
When try to start oscam on 64bit OS with x86_64 oscam with Softcam.key and oscam crash and show in console: double free or corruption (!prev).
This show only when use Softcam.key, when remove Softcam.key no problem.
Interesting is when use same Softcam.key on enigma2 don't have problem.

joeusercz, 2018-07-26:
I have also seen this, but only occasionally. I think it started when I configured two emu readers. Do you have two emu readers?

I have had oscam crash on startup a few times on enigma2 (sh4), but never logged it as it rarely happens. I will try setting up some logs.

lprot, 2018-07-26:
Hello, joeusercz! 

Can you drop me a mail at lprotasov@gmail.com? Have a question for you :)

Thx in advance.

oscam-emu, 2018-07-26:
There is an issue with oscam-emu crashing after an enigma2 reboot, when there are 2 emu readers configured. I've noticed this some time ago. I'ts probably related to the key deletion from memory.

The emu reader deletes all keys from memory when it starts up. This design was done with only 1 active emu reader in mind.  With 2 or more emu reader setups becoming common to overcome the powervu problems, this problem has raised.

But don't forget that having more than 1 emu reader configured is not supported and it is written in the wiki pages here. I maybe re-design the way the key deletion is done in the future though.

zinonino, 2018-07-26:
Yes, i use 2 emu readers due problem with AFN and other PowrVU channels, but problem not happend in enigma 2 ARM box don't crash, on PC where is more powerfull and 64 bit crash.

Saentist, 2018-07-26:
Do you try with less lines inside softcam.key

zinonino, 2018-07-26:
Saen acro a.k.a Joro,
First thing who came in my mind was to reduce number of keys, this is partly solution but what is effect where must put only 1-5  lines to not creash, i use AFN and Discovery on 1W, what left if all put PowerVu for other sat + BISS + Tandberg.

Yeah, problem come from  2 emu readers, but then what is other solution for AFN where all we know for  work AFN must use 2nd emu reader with cache enable.

Some other line who show me when trying to start oscam:
`oscam: malloc.c:2401: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.`

`corrupted size vs. prev_size`

Quick search in google say me: 99.9% likely that you have corrupted memory (over- or under-flowed a buffer, wrote to a pointer after it was freed, called free twice on the same pointer, etc.)

OK, problem with memory, but in enigma 2 box with more small memory, don't have problem strange.

dulcet, 2018-07-27:
Hello, zinonino
use multi oscam  1 for AFN and 2nd for Power Vu

jlmariense, 2018-09-27:
here in South America already two satellites have entered hash mode 17

nautilus7, 2018-11-01:
Try with [r775](https://github.com/oscam-emu/oscam-emu/commit/cedd73e4efed3625ecb1600a837e2ae2c8ade816) (which prevents parallel access to the key db and the key file) and report back if it is fixed.

oscam-emu, 2018-11-01:
Try with [r775](https://github.com/oscam-emu/oscam-emu/commit/cedd73e4efed3625ecb1600a837e2ae2c8ade816) (which prevents parallel access to the key db and the key file) and report back if it is fixed.

zinonino, 2018-11-02:
Confirm fixed, nice work @nautilus7 thanks.

oscam-emu, 2018-11-02:
Thanks for the report. I re-opened this because I found some other cases when parallel access to the key can happen that were missed with the previous fix.

I'll close it again when  I fix them as well. But nice to hear that is was a step in the right direction.

oscam-emu, 2018-11-07:
Added a restriction for those cases mentioned above in r776.

closed 2018-11-07 by oscam-emu