oscam-emu/oscam-emu #281: A new hash mode [17]

jlmariense, 2018-08-13:
2018/08/13 15:36:49 552101BD c      (ecm) vuplus-team.com (0E00@000000/0000/00A4/48:01C0C749115CA294E5AC3862405FC088): found (4 ms) by emulator (L/1/3/3) - Comutacao HD
2018/08/13 15:36:50 4D15FB67 r      (emu) A new hash mode [17] is in use.

jlmariense, 2018-10-13:
Ola amigo, passa teu zap, para podermos conversar melhor.

cooliobr, 2018-10-13:
48 991010802

Att.

Rafael Machado
REGS IT LTDA.
ANTIK Business Representative BRASIL
"Believe in the future, believe in your success"
"Não existe patch para a estupidez humana"

scolanzi, 2018-10-16:
Thanks to developers for the inclusion of hash modes 15 and 16 and look forward to mode 17.

oscam-emu, 2018-10-19:
added with r773 https://github.com/oscam-emu/oscam-emu/commit/47c4bd465b447862a83f380294f89455fbccc540

closed 2018-10-19 by oscam-emu

jlmariense, 2018-10-20:
Aki off na tp ESPN e Globosat. Estou vendo em fóruns que estão off tbm quem
está usando hash Mode 17